toblum/McLighting #90: esp01 compile errors

sander816, 2018-01-26:
when compiling on esp01:

`In file included from C:\Users\sande\Documents\Arduino\McLighting-master\Arduino\McLighting\McLighting.ino:1:0:

sketch\definitions.h:19:1: warning: narrowing conversion of '5.0e+0' from 'double' to 'uint32_t {aka unsigned int}' inside { } [-Wnarrowing]

 };

 ^

sketch\definitions.h:19:1: warning: narrowing conversion of '1.0e+1' from 'double' to 'uint32_t {aka unsigned int}' inside { } [-Wnarrowing]

sketch\definitions.h:19:1: warning: narrowing conversion of '5.0e+0' from 'double' to 'uint32_t {aka unsigned int}' inside { } [-Wnarrowing]

sketch\definitions.h:19:1: warning: narrowing conversion of '1.5e+1' from 'double' to 'uint32_t {aka unsigned int}' inside { } [-Wnarrowing]

C:\Users\sande\Documents\Arduino\McLighting-master\Arduino\McLighting\McLighting.ino: In function 'void writeEEPROM(int, int, String)':

C:\Users\sande\Documents\Arduino\McLighting-master\Arduino\McLighting\McLighting.ino:99:36: warning: passing NULL to non-pointer argument 2 of 'void EEPROMClass::write(int, uint8_t)' [-Wconversion-null]

       EEPROM.write(i + offset, NULL);

                                    ^

In file included from C:\Users\sande\Documents\Arduino\McLighting-master\Arduino\McLighting\McLighting.ino:160:0:

sketch\request_handlers.h: In function 'char* listStatusJSON()':

sketch\request_handlers.h:231:8: warning: address of local variable 'json' returned [-Wreturn-local-addr]

   char json[255];

        ^

sketch\request_handlers.h: In function 'void mqtt_callback(char*, byte*, unsigned int)':

sketch\request_handlers.h:465:19: warning: converting to non-pointer type 'uint8_t {aka unsigned char}' from NULL [-Wconversion-null]

   payload[length] = NULL;
`

toblum, 2018-01-26:
Hi @sander816,

I did not experience such behaviour before. What version of Arduino IDE and  ESP core do you use?

sander816, 2018-01-27:
updated everything, and its fine again. No idea
Other problem: when i upload i do net get a new wifi network, as set in definitions.h
Instead it just connects to my wifi network automatically without setting anything

toblum, 2018-01-27:
Hi @sander816,
if it can connect to the stored WLAN, it connects directly. But you could call http://<mc_lighting_name>/start_config_ap to start the config AP.

closed 2018-01-27 by toblum

myk3y, 2018-08-12:
I've just5 downloaded McLighting, on a freshly-installed 1.8.5 on 64-bit Mac OS with all new libraries installed and I'm getting the same message: 

In file included from /Users/mike/Documents/Arduino/McLighting/McLighting.ino:1:0:
sketch/definitions.h:43:1: warning: narrowing conversion of '5.0e+0' from 'double' to 'uint32_t {aka unsigned int}' inside { } [-Wnarrowing]
 };
 ^
sketch/definitions.h:43:1: warning: narrowing conversion of '1.0e+1' from 'double' to 'uint32_t {aka unsigned int}' inside { } [-Wnarrowing]
sketch/definitions.h:43:1: warning: narrowing conversion of '5.0e+0' from 'double' to 'uint32_t {aka unsigned int}' inside { } [-Wnarrowing]
sketch/definitions.h:43:1: warning: narrowing conversion of '1.5e+1' from 'double' to 'uint32_t {aka unsigned int}' inside { } [-Wnarrowing]

This is the code:

// parameters for automatically cycling favorite patterns
uint32_t autoParams[][4] = { // color, speed, mode, duration (seconds)
  {0xff0000, 200,  1,  5.0}, // blink red for 5 seconds
  {0x00ff00, 200,  3, 10.0}, // wipe green for 10 seconds
  {0x0000ff, 200, 11,  5.0}, // dual scan blue for 5 seconds
  {0x0000ff, 200, 42, 15.0}  // fireworks for 15 seconds
};

Changing the 5.0, 10.0, 5.0, 15.0 to 5, 10, 5, 15 respectively gets rid of the problem.

toblum, 2018-08-12:
Hi @myk3y,

what ESP core version are you using?

Tobias

toblum, 2018-08-20:
Yes, I meant the 2.4.2. I couldn't reproduce this on my system and it wasn't reported by someone else, so we'll keep an eye on it.

myk3y, 2018-08-20:
If there’s any other faultfinding I can do, let me know. I’ll do another compile with all debugging turned on and post to pastebin.

myk3y, 2018-08-24:
Interesting....

I just downloaded a fresh copy of McLighting, ran a compile on untouched base code and no errors...

Hmmm...

So I went back to my modified code, replace the '.0's at the end of the constants and it, too, compiled cleanly.

My memory isn't great, but I don't remember changing anything on Arduino IDE, but I may have updated a library or two between making the original post and now.

So, theres no problem - it was something on my Arduino setup that has been updated since the original post that was causing the issue. Sorry to trouble you.

processor83, 2018-10-19:
I have the same problème.

Changing the 5.0, 10.0, 5.0, 15.0 to 5, 10, 5, 15 respectively gets rid of the problem.

I used all version showed on this guide https://github.com/toblum/McLighting/wiki/Software-installation

can it be because i use french version of windows ?? Im happy that it now compile and i can test you software in the week-end 📦

myk3y, 2018-10-19:
No, it won’t have anything to do with your windows language settings.

I’m on a Mac with Australian English :)

toblum, 2018-10-20:
That is strange. I never experienced that. The compilation should be locale independent. I will keep an eye on it.