toblum/McLighting #93: WLAN integration, number of pixels & GPIO

wunderbaum, 2018-02-03:
Hello,
GREAT project! I am very happy with it! I am working with an Holmoe from ikea and a 

Here are my suggestions for improvement:
- The WLAN setup should be enhanced. I like to make a lamp as a gift and do not know the WLAN data where it is going to. This should be more flexible, maybe like in tasmota or other esp based projects. Some gpios are still free, so can connect a button or what else.
- It should be possible to configure the number of pixels via the web and the GPIO where the data pin is connected, too.
- OTA update for new versions is a great idea
- configuration for mqtt can be more flexible via web front.
- Uploading the webpage to spiffs is not easy for beginners, can this be included in the source?

So what I like to say, as much as possible should be configurable via web.

toblum, 2018-02-04:
Hi @wunderbaum,

thanks for your feedback. I added your suggestions to the list for future improvements. We will see if it may be implemented by me or another user in the future.

Regards
Tobias

closed 2018-02-04 by toblum

roesl, 2018-02-14:
I flashed the a nodemcu V0.9 with the firmware and loaded the index.html but I cant access the configuration of MQTT , pixel and pin configuration.
Please provide the correct index.html 

Is there a list of MQTT commands ?

thanks 
Georg

yusufsulaiman, 2018-04-08:
hi. i have a problem :
Coudnt find "/stripstate.json"
 Failure!
what can i do?

debsahu, 2018-04-08:
@yusufsulaiman it happens the first time you run McLighting, set a color or turn on the light. Once this happens it creates stripstate.json file.

Cifi76, 2018-07-05:
Hallo, also Ich habe das gleiche Problem 

Coudnt finde "/stripstate.json" 
Fehler! 

Die LEDs reagieren erst gar nicht.
Ich habe es wie folgt angeschlossen 

VIN = LED 5V
G    = LED -
D5  = Steuerkanal

Im Script sieht das so aus:

// Neopixel
#define PIN D5           // PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 10      // Number of leds in the strip
#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
#define BUTTON 4         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

Warum brennen die LEDs nicht ? 
Wo liegt der Fehler ?

Danke für eure Hilfe

The-Holgi, 2018-11-20:
Hi,
is there a way to edit the mqtt settings after the system is running in the home wlan?
My problem is, the mosqitto server run yet under different ip 
Thanks Holger

The-Holgi, 2018-11-20:
Hi, output after 
http://192.168.178.71/edit.htm
`File Not Found

URI: /edit.htm
Method: GET
Arguments: 0`

wunderbaum, 2018-11-20:
Pity. I thought thats possible that way.

debsahu, 2018-11-20:
If one doesn't upload edit.htm, you don't access to it. Anyways do a reset, http://<ipaddress>/reset_wlan

The-Holgi, 2018-11-20:
Thank‘s,
now it run‘s

jseadragon, 2019-06-26:
Hi @toblum, would this be something you'd be open to taking a PR for? I'm looking to deploy a handful of these for a personal project and it would be easier for end users (my family) if we could configure the # of pixels on the fly rather than at flash time. Specifically I could see about helping with that part if you are open to it.

I've also thought about OTA updates, but I don't think that's something I could tackle easily having not worked in this codebase much. Ideally you'd just `HTTP` `POST` the binary up to the ESP and it would store it, and then reboot.

jseadragon, 2019-06-26:
Actually, is the # of LED config option already done? https://github.com/toblum/McLighting/wiki/HTTP-REST-API#get-pixelconf-http10

What I'm hoping for is something that will persist after a reboot of the device.

toblum, 2019-06-26:
@jseadragon Yes, this sould be available if you use the latest version. The documentation needs to be updated accordingly. And yes, it is persisted.