toblum/McLighting #94: No issue at all, but ...

bpohvoodoo, 2018-02-08:
Hi,

i had made some modifications in my fork for supporting sk6812 and maybe other RGBW stripes, and regarding the status of the index.html after reloading the page.
Maybe someone wants have a look. @toblum Maybe you want to integrate in your project.

https://github.com/FabLab-Luenen/McLighting

Sincerely yours
Mirco

toblum, 2018-02-10:
Hi @bpohvoodoo,

thank you for your message. I will have a depper look at it as soon as I find time. I will have to see how I can make it work for RGB and RGBW as well.

Regards
Tobias

kotelmach, 2018-02-14:
I'd like to contribute with UX improvements, how can this be facilitated?

toblum, 2018-02-17:
Hi @kotelmach,

there is a src folder for the frontend files. It contains a gulp-script that packs up everything so that everything is contained in one file in the build folder.

Regards
Tobias

martinschki, 2018-04-07:
well, I did...
but I've no idea what to look for
can someone please point me to what to look for?

zavadetskiy, 2018-04-19:
hi
In file included from D:\McLighting-master\Arduino\McLighting\McLighting.ino:97:0:

C:\Users\Ігор\Documents\Arduino\libraries\WS2812FX/WS2812FX.h:161:37: error: 'neoPixelType' has not been declared

     WS2812FX(uint16_t n, uint8_t p, neoPixelType t) : Adafruit_NeoPixel(n, p, t) {

                                     ^

debsahu, 2018-04-19:
@zavadetskiy Update your Adafruit's Neopixel library to the latest version. https://github.com/adafruit/Adafruit_NeoPixel

zavadetskiy, 2018-04-19:
C:\Users\Ігор\Documents\Arduino\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp:364:46: error: 'wifi_station_get_reconnect_policy' was not declared in this scope

     return wifi_station_get_reconnect_policy();

                                              ^

C:\Users\Ігор\Documents\Arduino\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp:365:1: warning: control reaches end of non-void function [-Wreturn-type]

 }

JoJ123, 2018-10-14:
Could you please merge the latest master of the original repo into your fork?

JoJ123, 2018-10-14:
I adapted the McLightning adapter to support the RGBW Fork
https://github.com/JoJ123/ioBroker.mclighting/tree/feature/rgbwSupport

bpohvoodoo, 2019-02-03:
I have adapted the changes from 2.2.0 to the RGBW branch (https://github.com/FabLab-Luenen/McLighting).
This branch is now compatible with RGB and RGBW stripes
by setting "define RGBW" in definitions.h
additionally you can turn support for the "White slider" in my version of the new webinterface (McLightingUI) on or off in settings menu (https://github.com/FabLab-Luenen/McLightingUI).
Testing would be helpful.
By the way
@tablum: How do i build the meshy version?
Seems not to work with "npm run build".

ChristophA, 2019-02-03:
Thank you very much for updating the RGBW version! I'll run some test next week.

toblum, 2019-02-03:
@bpohvoodoo 
Great! Will try it out as soon as my RGBW strip arrives.

I think @debsahu can say something regarding building of MeshyMcLIghting.

bpohvoodoo, 2019-02-04:
@toblum: I ment the built of McLightingUI to add additional nodes. This is not working for me.
You can put in additional nodes, but it will allways send to the node it runs on.

ChristophA, 2019-02-09:
First test of your RGBW version and your re-written UI was very positiv! No issues identified so far. Will be interessting to see how stable this version is. I'll report...

closed 2019-03-27 by bpohvoodoo