toblum/McLighting #97: Webpage for MQTT , pixel and pin configuration not available

roesl, 2018-02-14:
I flashed the a nodemcu V0.9 with the firmware and loaded the index.html but I cant access the configuration of MQTT , pixel and pin configuration.
Please provide the correct index.html

Is there a list of MQTT commands ?

thanks
Georg

toblum, 2018-02-17:
Hi @roesl,

just have a look here: https://github.com/toblum/McLighting/wiki/MQTT-API

Number of pixels and pin can currently ony be configured in the code.

Regards
Tobias

closed 2018-02-17 by toblum

roesl, 2018-02-17:
here the text of your readme.md
I dont find anything of the features written in your text.

- [ ] Make number of pixels, MQTT and PIN configurable via front end [Issue](https://github.com/toblum/McLighting/issues/93)
 - [ ] OTA update [Issue](https://github.com/toblum/McLighting/issues/93)
 - [ ] Bundle webpages instead of SPIFFS [Issue](https://github.com/toblum/McLighting/issues/93)

toblum, 2018-02-17:
This is correct, as this is the todo list. :-)

roesl, 2018-02-18:
ok ! when will the features be realized ?

toblum, 2018-02-18:
when me or someone else finds time to implement it. There is no schedule.