toblum/McLighting #101: Change the NUMLEDS in the WebInterface?

Primus007, 2018-03-06:
can you add the possibility to change the Variable NUMLEDS in the Webinterface? Or can you tell me how to do this? 
I want to connect different stripes or led rings to my module. But everytime i must change the NUMLEDS, compile the scetch and upload it. it would be more comfortable if i can change it in the web interface.

thank you.

kotelmach, 2018-03-06:
I would recommend adding a settings menu, that way it can be hidden with custom CSS if needed.

toblum, 2018-03-07:
Hi @Primus007,

yes, this should be possible and already on the ToDo list (https://github.com/toblum/McLighting#todos). But this requires many changes (config savefile, API endpoints, UI) and needs some time to implement, that I did not find so far.

Regards
Tobias

closed 2018-04-07 by toblum