toblum/McLighting #108: MQTT

The-Holgi, 2018-03-18:
Hallo,
i don‘t understand how the mqtt topic must appearance.
Test with publish /McLighting01/in/=off dont work.
An example would fine.
Regards Holger

debsahu, 2018-03-18:
If your hostname is McLighting01, publish in this topic "McLighting01/in" without quotes. Message to publish is "=off" without quotes to turn off LEDs.

The-Holgi, 2018-03-18:
Thanks, it run´s, but not all lightmode´s. Fire Flicker dont starts over mqtt.
Is there a way to start the other lightmodes over mqtt ?

debsahu, 2018-03-18:
Send payload of "/48" without quotes to "McLighting01/in" topic should do the job. The payloads are same as described for websockets: https://github.com/toblum/McLighting/wiki/WebSocket-API

closed 2018-03-20 by debsahu