toblum/McLighting #111: Can you add a music visualizer to McLighting?

oldlyson, 2018-03-23:
I love this project so much
and your project is a my inspiration to do a arduino lerning
and I need some special feature
a music visualizer like this
https://www.youtube.com/watch?v=PBjSLQ1NhDk
https://www.youtube.com/watch?v=X3dh8sQKLXs

thank you so much :)

toblum, 2018-03-23:
Hi @oldlyson,

thank you for the good feedback. It's always good to hear.

The visualizer is also built on the Adafruit library, but for an arduino. It might be possible to port this to the ESP, but I'm afraid this would take too long, because I'm not so experienced with this.

The WS2812FX lib that is used by McLighting has a trigger function that could be used for sound-to-light. I did not find time to try it out yet. Mabe that could be a option for you. https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/external_trigger/external_trigger.ino

Regards
Tobias

debsahu, 2018-03-23:
See Hyperion project https://hyperion-project.org/ I2S + WiFi + driving LEDs is pushing the limits of ESP8266, use a $5 pi zero instead. That being said @cnlohr is pushing the limits of what this chip can do https://github.com/cnlohr/esp8266ws2812i2s

cnlohr, 2018-03-23:
I hope that was a vote for colorchord, not for esp8266ws2812i2s :-p.

schwma, 2018-03-26:
I would also like to mention this project that I find very interesting.
https://github.com/scottlawsonbc/audio-reactive-led-strip

toblum, 2018-03-27:
@schwma This is cool, but needs a separate PC or raspberry to work.

MrSnuggles, 2018-04-02:
take a look on this https://github.com/mtongnz/ESP8266_ArtNetNode_v2 with a software like jinx or a DMX input you can do a lot with music.

toblum, 2018-04-03:
There is also a (older) McLighting version with Artnet support available: https://github.com/toblum/McLighting/tree/mclighting_artnet

closed 2018-04-07 by toblum

toblum, 2018-04-07:
I added this to list list for possible enhancements.

schnellecom, 2018-04-21:
I would also really enjoy the option. 

BTW awesome project keep up the good work

processor83, 2018-10-19:
i would like to try the old version with artnet what is different ?

toblum, 2018-10-20:
@processor83 The artnet version is quite old. It misses some features and you may use an older version of the arduino core and libraries.

Regards
Tobias

Pejosonic, 2019-03-31:
I'been McLighting v2 for more than a year. I'm going to test the artnet version. 
It's possible to implement artnet in current version?

Thanks!

debsahu, 2019-03-31:
It now supports E1.31 SACN protocol.

Pejosonic, 2019-03-31:
This is a very good news! So it's supposed to be able to control from Jinx. 
When was added sACN support to McLighting?