toblum/McLighting #113: Latest WS2812FX Breaks McLighting?

ke5, 2018-03-28:
I just updated WS2812FX (downloaded the .zip from github) and rebuilt the latest McLighting master branch. All appeared to work correctly until I tried to load the web interface. It just hangs, looks like it's trying to read the list of modes. WS2812FX has recently added an extra 'custom' mode, could this be the problem? Anyone else experiencing the same? I've tried various configurations (NodeMCU, Wemos D1 Mini) and numbers of LEDS.
Am I missing something or, more likely, done something foolish?

ke5, 2018-03-28:
I should add, I reverted to the previous WS2812FX library (thank goodness I save it!) and all works perfectly.

debsahu, 2018-03-30:
I used latest WS2812FX, did not notice any issues.

ke5, 2018-03-30:
I did a complete reinstall of the IDE, ESP8266 core and libraries, rebuilt and now everything is working nominally. Clearly, the problem was with my environment and not with WS2812FX or McLighting. Thanks for your help, I appreciate you taking the time to look into it.

closed 2018-03-30 by ke5