toblum/McLighting #117: Continous loop, webinterface unresponsive

MrSnuggles, 2018-04-02:
I flashed my wemos d1 mini with the nodemcu software. Now I have one effect in a loop and can't chose a color in the webinterface (loading forever). Should I reflash it with the wemos mini boardsettings?

> SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)`
readEEPROM(): 1
readEEPROM(): 
readEEPROM(): 
readEEPROM(): 
readEEPROM(): 
MQTT host: 
MQTT port: 
MQTT user: 
MQTT pass: 
*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
connected with XXXXX, channel 1
dhcp client start...
ip:192.168.2.127,mask:255.255.255.0,gw:192.168.2.1
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.2.127
connected...yeey :)
Arduino OTA activated.
Open http://192.168.2.127/ to open McLighting.
Use http://McLighting01.local/ when you have Bonjour installed.
New users: Open http://192.168.2.127/upload to upload the webpages first.
[WS-Server] Server Started.
FS File: /, size: 0B
FS File: /index.htm, size: 13.90KB
FS Usage: 14809/957314 bytes
readEEPROM(): STA| 1|  7|155| 45|255| 43|  0
Found saved state: STA| 1|  7|155| 45|255| 43|  0
ws2812fx_mode: 7
ws2812fx_speed: 155
brightness: 45
main_color.red: 255
main_color.green: 43
main_color.blue: 0
*WM: freeing allocated params!
pm open,type:2 0

toblum, 2018-04-03:
Hi @MrSnuggles,

I would definitely give it a try. The wemos baords are always a bit different.

Regards
Tobias

LeonVos, 2018-04-03:
Maybe you did forget to upload the webpages ??

toblum, 2018-04-03:
According to the debug output a index.htm is present. From which device are you trying to load the UI? If from desktop, could you open the developer extension (usually F12) and see if there are any aerrors logged in the console?

toblum, 2018-04-07:
@MrSnuggles Any news on this?

closed 2018-04-07 by toblum