toblum/McLighting #118: Feature Request: Multiple LED strips/data outputs

cyberjacob, 2018-04-04:
Would be useful if a single NodeMCU could be used to run multiple LED strips on different pins. They would need to be able to use different configurations as well to be useful (i.e. one strip on red, one on green)

debsahu, 2018-04-04:
That's pushing ESP8266 too hard, may not be stable. I haven't tried it, but it can be done by defining multiple WS28xx strips via WS2812FX.

Another way to approach this is to use only one pin, create segments and daisy chain all your WS28xx strips. McLighting currently doesnt support segments (but the framework of segments are present in WS2812FX or NeoAnimationFX (beta) that is used by McLighting), but may come in later. Maybe as a separate branch.

Since these chips cost <$3, why not use one for every strip. If you want them coordinated, you can let them subscribe to same MQTT topic. This can be done by giving same hostnames and use different MQTT hostname IDs.

cyberjacob, 2018-04-04:
Only potential issue with segments would be that the data cabling needs to go from the end of one strip to the start of another. Downside to using multiple ESPs would be that any external components required to run them (e.g. 12v to 5v micro USB adapter) would need to be duplicated, pushing the cost price up further.

If this is a processing power limitation on the ESP8266, there's not much that can be done about it. I'll see if I can do some testing with the strips I've already got and check how the ESP responds.

toblum, 2018-04-07:
I've put the on the list for possible enhancements. But as @debsahu already said: this is limited by the capabilities of the ESP8266.
Maybe I could find at least a way to control multiple McLightings from one webinterface.

closed 2018-04-07 by toblum

cyberjacob, 2018-04-09:
I think controlling multiple instances from one interface is already handled by MQTT

toblum, 2018-04-09:
Yes, I use it also this way, but not everyone has a homeserver running.

myk3y, 2018-12-17:
Have you considered using fastLED library? It handles multiple strips, many different types of LED and has a huge user base and associated patterns.

https://github.com/FastLED/FastLED