toblum/McLighting #119: Feature Request: Multiple buttons/GPIO Inputs

cyberjacob, 2018-04-04:
Multiple buttons (or any form of GPIO input) that would feed back to MQTT, so as to trigger automations.

toblum, 2018-04-07:
@cyberjacob I added this to the list for possible enhancemants.

closed 2018-04-07 by toblum

JPluess, 2019-02-01:
hi i have not much experiance programming but i wrote a class for multibutton. would that be appreciated? 
it registers faspress and counts how many there are and retutns that, and it counts up if you hold the button down. every 0.5s it adds 1. 

#define BUTTON_PIN D3 // Button Pin on node mcu it is D3
#define TIME_GAP_SHORT 300 // Set the time between fast presses
#define TIME_GAP_LONG 500 // Set the time between long presses
#define BUTTON_STATE 0// Set the button to 1 HIGH on or  0 LOW on

MultiButton Button1(BUTTON_PIN, TIME_GAP_SHORT, TIME_GAP_LONG, BUTTON_STATE);

	if(uint8_t counts = Button1.ShortPress())
	{ 
		if (counts == 1) your code here;
		if(counts == 2) your code here;
		if(counts == 3) your code here;
		if(counts == 4) your code here;
	}

or the long press 
	if (uint8_t counts = Button1.LongPress())
	{
		if (counts == 1) your code here;
		if(counts == 2) your code here;
		if(counts == 3) your code here;
		if(counts == 4) your code here;
	}
Button1.fast_press_toggle // true or false is the debounced toggle of the specified button