toblum/McLighting #120: Change of speed -> no MQtt information about it

chkbln, 2018-04-04:
Changing Speed does not generate an MQtt event.
The speed information is also not included
{"state": "ON", "color": {"r": 217, "g": 121, "b": 79}, "brightness": 59, "effect": "Single Dynamic"}
Or is that intended?

Kind regards

debsahu, 2018-04-04:
It is intentional, there is no speed component in Home Assistant's MQTT JSON Light ( https://www.home-assistant.io/components/light.mqtt_json/ ). If you see my settings, I use a slider (input_number).

chkbln, 2018-04-04:
Where can I find your settings?

debsahu, 2018-04-04:
Currently it is one way, HA sends speed to McLighting. What you are looking for is McLighting send back speed to MQTT -> HA automation to read MQTT value and change HA input_slider. That can be done.

chkbln, 2018-04-04:
I agree. You have recognized my problem. I think it makes sense that homeatomation knows the state of McLighting.
Thank you for trying to find a solution.

debsahu, 2018-04-04:
Can you try my branch for testing ( https://github.com/debsahu/McLighting ), I confirmed it works on the McLighting side. But havent tested automation. Could you check and report back?

```
  - id: 93786598732698756967
    alias: NeoPixel Animation Speed Receive
    trigger:
    - platform: mqtt
      topic: home/McLighting01_ha/state/out
    action:
    - data:
        entity_id: input_number.neopixel_animation_speed
        value: '{{ trigger.payload | value_json.speed }}'
      service: input_number.set_value
```

debsahu, 2018-04-05:
I'm closing this as the new update on my branch should solve this. I removed the need for a series mqtt speed topic for HA. This will be eventually merged after @toblum approves.

closed 2018-04-05 by debsahu

chkbln, 2018-04-05:
Hello debsaha
Unfortunately I can not compile your branch. I get the error message "request_handlers.h: 407: error: 'class ticker' has no member named 'active'".
I use Arduino IDE 1.85 with Core 2.41.
Do you have an idea why this could be?
Best regards

debsahu, 2018-04-05:
I an confirm that this https://github.com/esp8266/Arduino/pull/2722/files was pushed before 2.4.1. I checked this on my VM (fresh install). Compiles fine on my 2.4.1 version and git version of esp8266 arduino. Could you reinstall Arduino IDE and/or esp8266 arduino 2.4.1?

chkbln, 2018-04-06:
Hello debsaha, 
I could now compile your sketch. I had to update some libraries. they still worked under the sketch of toblum. 
Compiling with AMQTT works. 
MQTT not. (error: no matching function for call to 'PubSubClient :: subscribe (char [23], uint8_t &, uint8_t &)')
If the button (// # define BUTTON 0) is commented out, I also get an error message (error: 'BUTTON' was not declared in this scope)

When you click on TV
[image]
 , the MQTT event does not show the TV modes.

debsahu, 2018-04-06:
> MQTT not. (error: no matching function for call to 'PubSubClient :: subscribe (char [23], uint8_t &, uint8_t &)')

Install [PubSubClient](https://github.com/knolleary/pubsubclient)

> If the button (// # define BUTTON 0) is commented out, I also get an error message (error: 'BUTTON' was not declared in this scope)

Do not comment that line. Comment this line instead `// #define ENABLE_BUTTON`

> When you click on TV, the MQTT event does not show the TV modes.

It was written way before I started I was involved in this project, it is not a WS2812FX animation. @toblum hard coded this, so can't be used by MQTT.