toblum/McLighting #121: Feature Request: update the Homeassistant Config, add Auto Discovery

TechOnlyGamer, 2018-04-04:
Hey there iam fairly new to programming with only a few basics. But i hope to see ur Light.yaml file updated with ur latest release cause mine stopped working after updatind trying to test the asyncmqtt. Cause my Light doesnt work if my Hassio is restarzing/updating. I have to manually restart it everytime. Second thing i thought about is adding the mqtt discovery option to the esp. As i can see from my research there has to be a few lines of code creating a json with the settings foe the yaml file. publishing everytime in the reconnect function. Maybe u can have a look at it too. Greetings from Germany and thanks for ur hard work all together

debsahu, 2018-04-04:
Light.yaml remains the same. What might be happening is that when HASSIO restarts, it also restarts the MQTT server hosted there. So AMQTT/PubSubClient retries re-connection every 5s or so and gives up after 3 tries.

Regarding creating a discovery component, highly unlikely to create using ESP8266 as it would require one to supply every setting in yaml file in JSON format. That is way too long message to be sent via MQTT. The limit set here is 512 bytes for AsyncMQTT and PubSubClient.

TechOnlyGamer, 2018-04-04:
i have seen something like this working. but in this case the effect_list array only contains a rainbow entry. dont know how much effects we could add to the list wothout a problem but would male simple Installation with less pre defined effects waaaay faster.

TechOnlyGamer, 2018-04-04:
if defined(MQTT_HOME_ASSISTANT_SUPPORT)

// template: <discovery prefix>/light/<chip ID>/config, status, state or set

#define MQTT_CONFIG_TOPIC_TEMPLATE "%s/light/%s/config"

#else

TechOnlyGamer, 2018-04-04:
only thing i struggle and cause i cant test it myself i dont understand the config topic variable with the %s... dont know what that is used for and didnt find any Information about it...

debsahu, 2018-04-04:
Async MQTT works with mosquitto without issues. Async issues might be HASSIO specific, which not always translates to Home Assistant and/or mosquitto. They are different beasts.

Assuming you want to send this
```
{
  "name": "McLighting01",
  "platform": "mqtt_json",
  "state_topic": "jsbf;kjbsdkjlfbsdkjlfkdsjlfkjlsdhfkj",
  "command_topic": "slkfhlsdkjhfkjlsdhlkds",
  "on_command_type": "first",
  "brightness": "true",
  "rgb": "true",
  "optimistic": "false",
  "color_temp": "true",
  "effect": "true",
  "effect_list": ["Static", "Rainbow"]
}
```

Could you add this to `mqtt_reconnect()` inside `mqtt_client.connect()` ?
```
          #define MQTT_HOME_ASSISTANT_SUPPORT
          #ifdef MQTT_HOME_ASSISTANT_SUPPORT
            DynamicJsonBuffer jsonBuffer(JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(11));
            JsonObject& json = jsonBuffer.createObject();
            json["name"] = HOSTNAME;
            json["platform"] = "mqtt_json";
            json["state_topic"] = mqtt_ha_state_out;
            json["command_topic"] = mqtt_ha_state_in;
            json["on_command_type"] = "first";
            json["brightness"] = "true";
            json["rgb"] = "true";
            json["optimistic"] = "false";
            json["color_temp"] = "true";
            json["effect"] = "true";
            json["effect_list"] = "[\"Static\", \"Rainbow\"]";
            char buffer[json.measureLength() + 1];
            json.printTo(buffer, sizeof(buffer));
            mqtt_client.publish(String("homeassistant/light/" + String(HOSTNAME) + "/config").c_str(), buffer, true);
          #endif
```

TechOnlyGamer, 2018-04-06:
tested it about 15 mins ago. doesnt appear in homeassistant. dont know how to log a potantial mistake. But got async working now xD

debsahu, 2018-04-06:
Did you add
```
mqtt:
  discovery: true
  discovery_prefix: homeassistant
```
to your configuration.yaml?

debsahu, 2018-04-07:
@toblum This will not be integrated into the main branch, just keeping this open to help @TechOnlyGamer out.

toblum, 2018-04-07:
@debsahu OK, good to know.

debsahu, 2018-04-08:
@TechOnlyGamer Unlike a switch, this has to send a lot of data over MQTT in one (config) message. Unfortunately, there is a limitation of 512 bytes that can be sent in one go using PubSubClient. I dont want HA to be able to use this with limited animations. That is why I said this will not be merged into master.

Things I did:
1. HA Settings: 
```
mqtt:
  broker: 192.168.0.xxx
  port: 1883
  client_id: home-assistant
  keepalive: 60
  discovery: true
  discovery_prefix: homeassistant
```
Use MQTTfx to send this data to my MQTT server.
```

{
  "name": "McLighting01",
  "platform": "mqtt_json",
  "state_topic": "home/McLighting04_ha/out",
  "command_topic": "home/McLighting04_ha/in",
  "on_command_type": "first",
  "brightness": "true",
  "rgb": "true",
  "optimistic": "false",
  "color_temp": "true",
  "effect": "true",
  "effect_list": ["Static", "Rainbow"]
}
```
Restart HA and the light shows up on my HA.
[image]

2. Inserted following lines (**different from what I posted before**) in `mqtt_client.connect()`, compiled and uploaded the code
```
          #define MQTT_HOME_ASSISTANT_SUPPORT
          #ifdef MQTT_HOME_ASSISTANT_SUPPORT
            DynamicJsonBuffer jsonBuffer(JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(11));
            JsonObject& json = jsonBuffer.createObject();
            json["name"] = HOSTNAME;
            json["platform"] = "mqtt_json";
            json["state_topic"] = mqtt_ha_state_out;
            json["command_topic"] = mqtt_ha_state_in;
            json["on_command_type"] = "first";
            json["brightness"] = "true";
            json["rgb"] = "true";
            json["optimistic"] = "false";
            json["color_temp"] = "true";
            json["effect"] = "true";
            JsonArray& effect_list = json.createNestedArray("effect_list");
            effect_list.add("Static");
            effect_list.add("Rainbow");
            char buffer[json.measureLength() + 1];
            json.printTo(buffer, sizeof(buffer));
            mqtt_client.publish(String("homeassistant/light/" + String(HOSTNAME) + "/config").c_str(), buffer, true);
          #endif
```
Subscribed to `home/McLighting04_ha/#` on MQTTfx and get the following message

`{"name":"McLighting04","platform":"mqtt_json","state_topic":"home/McLighting04_ha/state/out","command_topic":"home/McLighting04_ha/state/in","on_command_type":"first","brightness":"true","rgb":"true","optimistic":"false","color_temp":"true","effect":"true","effect_list":["Static","Rainbow"]}`

Restart HA and the light shows up on my HA as well.
[image]

As you can see the message that is sent is already 292 bytes, so every effect that WS2812FX offers can't be put in!

debsahu, 2018-04-08:
When I try to send everything using the following code:
```
          #define MQTT_HOME_ASSISTANT_SUPPORT
          #ifdef MQTT_HOME_ASSISTANT_SUPPORT
            DynamicJsonBuffer jsonBuffer(JSON_ARRAY_SIZE(strip.getModeCount()) + JSON_OBJECT_SIZE(11));
            JsonObject& json = jsonBuffer.createObject();
            json["name"] = HOSTNAME;
            json["platform"] = "mqtt_json";
            json["state_topic"] = mqtt_ha_state_out;
            json["command_topic"] = mqtt_ha_state_in;
            json["on_command_type"] = "first";
            json["brightness"] = "true";
            json["rgb"] = "true";
            json["optimistic"] = "false";
            json["color_temp"] = "true";
            json["effect"] = "true";
            JsonArray& effect_list = json.createNestedArray("effect_list");
            for (uint8_t i = 0; i < strip.getModeCount(); i++) {
              effect_list.add(strip.getModeName(i));
            }
            char buffer[json.measureLength() + 1];
            json.printTo(buffer, sizeof(buffer));
            mqtt_client.publish(String("homeassistant/light/" + String(HOSTNAME) + "/config").c_str(), buffer, true);
          #endif
```
Total size is 1.9kB. MQTT cant handle it.

That being said, when I compile using AMQTT (now the default for McLighting) on 2.4.1 arduino esp8266, Tools -> IwIP variant -> Higher Bandwidth, I am able to send 1.8kB TCP message
[image]

`{"name":"McLighting04","platform":"mqtt_json","state_topic":"home/McLighting04_ha/state/out","command_topic":"home/McLighting04_ha/state/in","on_command_type":"first","brightness":"true","rgb":"true","optimistic":"false","color_temp":"true","effect":"true","effect_list":["Static","Blink","Breath","Color Wipe","Color Wipe Inverse","Color Wipe Reverse","Color Wipe Reverse Inverse","Color Wipe Random","Random Color","Single Dynamic","Multi Dynamic","Rainbow","Rainbow Cycle","Scan","Dual Scan","Fade","Theater Chase","Theater Chase Rainbow","Running Lights","Twinkle","Twinkle Random","Twinkle Fade","Twinkle Fade Random","Sparkle","Flash Sparkle","Hyper Sparkle","Strobe","Strobe Rainbow","Multi Strobe","Blink Rainbow","Chase White","Chase Color","Chase Random","Chase Rainbow","Chase Flash","Chase Flash Random","Chase Rainbow White","Chase Blackout","Chase Blackout Rainbow","Color Sweep Random","Running Color","Running Red Blue","Running Random","Larson Scanner","Comet","Fireworks","Fireworks Random","Merry Christmas","Fire Flicker","Fire Flicker (soft)","Fire Flicker (intense)","Circus Combustus","Halloween","Bicolor Chase","Tricolor Chase","ICU","Custom","Meteor Rain"]}`

So it is possible to set it up, but lot of hassle on the user-side to be able to compile for a certain IwIP variant.

debsahu, 2018-04-08:
@toblum Now that AMQTT is the default MQTT client, one has to select a Higher Bandwidth IwIP variant to be able to use MQTT discovery for HA. This will mean that there is zero configuration involved in setting up the light on HA (except for speed). My previous statement about not merging with master is now **invalid**! So it is up to you, do you want merge https://github.com/debsahu/McLighting/tree/ha-mqtt-discovery ? 

@TechOnlyGamer Could you try my [branch](https://github.com/debsahu/McLighting/tree/ha-mqtt-discovery) and see if you are able to reproduce my results?

toblum, 2018-04-08:
Hi @debsahu and @TechOnlyGamer,

I can't completely assess the implications as I'm no homeassistant user and the "Higher Bandwidth IwIP" is also new to me. So I'll leave it up to you @debsahu if you're fine with it. 

Regarding the "Higher Bandwidth IwIP": This is only relevant if you want to use autodiscovery if I'm right. At least both settings work for me if using AMQTT, But I think we should mention this in the installation notes for all users. Right?

TechOnlyGamer, 2018-04-08:
yeah... sucess on my side tooo thank u very much its working like a charm... flashed it and after the reboot it showed up instantly on my ha instance thank u. i really like working things that doesnt clumb up my interface if not in use and is most likely plug and play like this now.

debsahu, 2018-04-08:
@toblum The improvements mentioned here are just an addition to autodiscovery of McLighting in HA. Plug and play. It sends the complete light configuration (light.yaml excluding speed control) to MQTT server.

There should be no compile errors in either IwIP settings, only thing is Max TCP message size is capped to 1kB for lower bandwidth and 4 kB for higher bandwidth. AMQTT can send 3-4kB data easily if TCP settings permit it. Yes if we merge this, we need to mention the IwIP setting in installation notes.

toblum, 2018-04-08:
Hi,

I merged the "ha-mqtt-discovery" branch now and updated documentation. I left "MQTT_HOME_ASSISTANT_SUPPORT" disabled by default for now.

Best regards
Tobias