toblum/McLighting #122: call for help: rgbW

martinschki, 2018-04-06:
Hi there!!
First of all, thank you for this amazing program. I had it running together with my MQTT server literally in minutes...

Heres my plan. I'm making a "cloud lamp" for my sons room with SK6812 which will do three things.
1. simulate the weather forecast for the next day: I will use node-red for that and depending on the forecast light the lamp
2. use the "W" in RGBW as a normal room lamp
3. use the "W" as a wake up light in the morning by dimming the "W" to max over a period of time. Also for this I will use node red via MQTT and dim the lamp

As far as I understand Tobias, it is difficult to implement the fourth color into the code. So here's my question, because, everything except the "W" works:

Is there a simple way, or a "hack" how I can send brightness to the 4th LED??
I'm using the "NEO_GRBW" setting already which works perfect in RGB.

Any help on how to insert a new MQTT command which turns on all the "W"s with the sent brightness would be appreciated...
Thank you!
Martin

btw. I got the inspiration for the project here:
https://www.youtube.com/watch?v=ckxU3XCHLoU
There are already a lot of variants floating around but none would be as elegant as this one..... :)

debsahu, 2018-04-06:
Please see https://github.com/toblum/McLighting/issues/94

martinschki, 2018-04-07:
i did....
but not being the uber programmer, I'd need someone to point me in the right direction...please...

toblum, 2018-04-07:
@martinschki The McLighting version thats mentioned in #94 supports RGBW. You could just take this version to use your RGBW strip: https://github.com/FabLab-Luenen/McLighting

martinschki, 2018-04-07:
i tried that but there are things not like in the latest master (MQTT is different, the blue led is always on?)
Is there a way to sync #94 to the latest master??

toblum, 2018-04-07:
The RGBW version has many changes in the code that would make problems for other users. That's the reason why I didn't integrate it so far. It would take a lot of rework and could break existing integrations.

bpohvoodoo, 2018-04-11:
All changes until today from this repository have been applied.

debsahu, 2018-04-12:
@bpohvoodoo NeoAnimationFX does not support RGBW as of now, so you could remove the NeoAnimationFX code. Since neither I or @toblum have this kind of light strips, support may be far away. Replacing RgbColor to RgbwColor should be able to convert most of the animations to RGBW format, maybe I will create a separate branch (no promises).

toblum, 2018-04-13:
I have one of those SKsomething strips from china lying here around, but I never got it to work (only as RGB). Maybe this was a fake product. Maybe I will buy another one in the future and give it another try.

bpohvoodoo, 2018-04-14:
Maybe i will have a look at your Lib and convert it, too.

martinschki, 2018-04-14:
@bpohvoodoo work in progress?
does need a couple of extra pics and not I'm stuck...

```
Arduino: 1.8.5 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Arduino/McLighting-master-2/Arduino/McLighting/McLighting.ino: In function 'void setup()':
McLighting:872: error: 'tcs' was not declared in this scope
   tcs.setConfig(MCU_LED_05, MCU_WHITE_ON);
   ^
McLighting:872: error: 'MCU_LED_05' was not declared in this scope
   tcs.setConfig(MCU_LED_05, MCU_WHITE_ON);
                 ^
McLighting:872: error: 'MCU_WHITE_ON' was not declared in this scope
   tcs.setConfig(MCU_LED_05, MCU_WHITE_ON);
                             ^
exit status 1
'tcs' was not declared in this scope

```

bpohvoodoo, 2018-04-15:
replace
tcs.setConfig(MCU_LED_05, MCU_WHITE_ON); 

with
#ifdef ENABLE_BUTTON_GY33
  tcs.setConfig(MCU_LED_05, MCU_WHITE_ON);
#endif

martinschki, 2018-04-15:
did!
it's now building and working!
(the repo https://github.com/FabLab-Luenen/McLighting doesn't have the changes in...)

two follow ups:
- how do I turn off the blue LED on the board (it's not on in Tobias' version)
- how do I control the white LED via MQTT?

Thanks!

bpohvoodoo, 2018-04-15:
> two follow ups:
> how do I turn off the blue LED on the board (it's not on in Tobias' version)
Are you sure? I haven't changed anything regarding the BUILTIN_LED.
> how do I control the white LED via MQTT?
Haven't set up a mqtt server yet, but you should send color as
FF000000 for white
00FF0000 for red
0000FF00 for green
and
000000FF for blue

martinschki, 2018-04-15:
thanks, white is working via MQTT as you said!

as for the blue LED: its on in your fork and off in the orig...
(i have a LoLin NodeMCU V3)

JoJ123, 2018-10-14:
I adapted the McLightning adapter to support the RGBW Fork
https://github.com/JoJ123/ioBroker.mclighting/tree/feature/rgbwSupport

JoJ123, 2018-10-15:
I think they said, there are too many  changes and differences between the codes

toblum, 2018-12-16:
Closing issue due to inactivity.

closed 2018-12-16 by toblum