toblum/McLighting #125: Latest .zip (commit 2256522) won't compile.

ke5, 2018-04-08:
Hi.
Just downloaded the latest .zip (commit 2256522 ) and now I can't get McLighting to compile. I'm getting multiple error messages (below), all reporting 'class Ticker' has no member named 'active'.
I've tried with NodeMCU and Wemos D1 Mini.
Please advise.
Kevin

Error log follows --->
Arduino: 1.8.5 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Build options changed, rebuilding all
In file included from /Users/kev/Documents/Arduino/McLighting/Arduino/McLighting/McLighting.ino:206:0:
sketch/request_handlers.h: In function 'void webSocketEvent(uint8_t, WStype_t, uint8_t*, size_t)':
request_handlers.h:404: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:419: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:444: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:463: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:522: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:578: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
sketch/request_handlers.h: In function 'void mqtt_callback(char*, byte*, unsigned int)':
request_handlers.h:807: error: 'class Ticker' has no member named 'active'
         if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                          ^
request_handlers.h:831: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:845: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:869: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:888: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:942: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
request_handlers.h:993: error: 'class Ticker' has no member named 'active'
           if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                            ^
/Users/kev/Documents/Arduino/McLighting/Arduino/McLighting/McLighting.ino: In lambda function:
McLighting:540: error: 'class Ticker' has no member named 'active'
       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                        ^
/Users/kev/Documents/Arduino/McLighting/Arduino/McLighting/McLighting.ino: In lambda function:
McLighting:567: error: 'class Ticker' has no member named 'active'
         if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                          ^
/Users/kev/Documents/Arduino/McLighting/Arduino/McLighting/McLighting.ino: In lambda function:
McLighting:785: error: 'class Ticker' has no member named 'active'
       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
                        ^
Multiple libraries were found for "ESP8266WebServer.h"
 Used: /Users/kev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266WebServer
 Not used: /Users/kev/Documents/Arduino/libraries/WebServer_tng
exit status 1
'class Ticker' has no member named 'active'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
<--- Error log ends

ke5, 2018-04-08:
Ok, I jumped the gun a bit - sorry!
I deactivated Home Assistant and now can successfully compile. Maybe Home Assistant should not be defined by default to avoid fools like myself having problems :)
Again, sorry about that (and as always great project, you guys do amazing work). Thanks.

toblum, 2018-04-08:
Hi @ke5,

which version of the ESP core are you using (boards manager in the Arduino IDE)? Latest version needs the 2.4.1 core. I have to add that in the documentation. The error message could be related to that.

Regards
Tobias

yanbec, 2018-04-08:
Updating ESP core from 2.4.0 to 2.4.1 fixed it for me :+1:

ke5, 2018-04-08:
I was on ESP Core 2.4.0, updated to 2.4.1 but it still fails to compile if I have ENABLE_STATE_SAVE_SPIFFS and works with ENABLE_STATE_SAVE_EEPROM. However, the error message is different, 'DynamicJsonBuffer' was not declared in this scope, log below.

Begin error log --->
Arduino: 1.8.5 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from /Users/kev/Documents/Arduino/McLighting/Arduino/McLighting/McLighting.ino:206:0:
sketch/request_handlers.h: In function 'bool writeConfigFS(bool)':
request_handlers.h:1222: error: 'DynamicJsonBuffer' was not declared in this scope
     DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(4));
     ^
request_handlers.h:1222: error: expected ';' before 'jsonBuffer'
     DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(4));
                       ^
request_handlers.h:1224: error: 'JsonObject' was not declared in this scope
     JsonObject& json = jsonBuffer.createObject();
     ^
request_handlers.h:1224: error: 'json' was not declared in this scope
     JsonObject& json = jsonBuffer.createObject();
                 ^
request_handlers.h:1224: error: 'jsonBuffer' was not declared in this scope
     JsonObject& json = jsonBuffer.createObject();
                        ^
request_handlers.h:1225: error: 'mqtt_host' was not declared in this scope
     json["mqtt_host"] = mqtt_host;
                         ^
request_handlers.h:1226: error: 'mqtt_port' was not declared in this scope
     json["mqtt_port"] = mqtt_port;
                         ^
request_handlers.h:1227: error: 'mqtt_user' was not declared in this scope
     json["mqtt_user"] = mqtt_user;
                         ^
request_handlers.h:1228: error: 'mqtt_pass' was not declared in this scope
     json["mqtt_pass"] = mqtt_pass;
                         ^
sketch/request_handlers.h: In function 'bool readConfigFS()':
request_handlers.h:1259: error: 'DynamicJsonBuffer' was not declared in this scope
       DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(4)+300);
       ^
request_handlers.h:1259: error: expected ';' before 'jsonBuffer'
       DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(4)+300);
                         ^
request_handlers.h:1261: error: 'JsonObject' was not declared in this scope
       JsonObject& json = jsonBuffer.parseObject(buf.get());
       ^
request_handlers.h:1261: error: 'json' was not declared in this scope
       JsonObject& json = jsonBuffer.parseObject(buf.get());
                   ^
request_handlers.h:1261: error: 'jsonBuffer' was not declared in this scope
       JsonObject& json = jsonBuffer.parseObject(buf.get());
                          ^
request_handlers.h:1266: error: 'mqtt_host' was not declared in this scope
         strcpy(mqtt_host, json["mqtt_host"]);
                ^
request_handlers.h:1267: error: 'mqtt_port' was not declared in this scope
         strcpy(mqtt_port, json["mqtt_port"]);
                ^
request_handlers.h:1268: error: 'mqtt_user' was not declared in this scope
         strcpy(mqtt_user, json["mqtt_user"]);
                ^
request_handlers.h:1269: error: 'mqtt_pass' was not declared in this scope
         strcpy(mqtt_pass, json["mqtt_pass"]);
                ^
sketch/request_handlers.h: In function 'bool writeStateFS()':
request_handlers.h:1290: error: 'DynamicJsonBuffer' was not declared in this scope
   DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(7));
   ^
request_handlers.h:1290: error: expected ';' before 'jsonBuffer'
   DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(7));
                     ^
request_handlers.h:1292: error: 'JsonObject' was not declared in this scope
   JsonObject& json = jsonBuffer.createObject();
   ^
request_handlers.h:1292: error: 'json' was not declared in this scope
   JsonObject& json = jsonBuffer.createObject();
               ^
request_handlers.h:1292: error: 'jsonBuffer' was not declared in this scope
   JsonObject& json = jsonBuffer.createObject();
                      ^
sketch/request_handlers.h: In function 'bool readStateFS()':
request_handlers.h:1333: error: 'DynamicJsonBuffer' was not declared in this scope
       DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(7)+200);
       ^
request_handlers.h:1333: error: expected ';' before 'jsonBuffer'
       DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(7)+200);
                         ^
request_handlers.h:1335: error: 'JsonObject' was not declared in this scope
       JsonObject& json = jsonBuffer.parseObject(buf.get());
       ^
request_handlers.h:1335: error: 'json' was not declared in this scope
       JsonObject& json = jsonBuffer.parseObject(buf.get());
                   ^
request_handlers.h:1335: error: 'jsonBuffer' was not declared in this scope
       JsonObject& json = jsonBuffer.parseObject(buf.get());
                          ^
Multiple libraries were found for "ESP8266WebServer.h"
 Used: /Users/kev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer
 Not used: /Users/kev/Documents/Arduino/libraries/WebServer_tng
exit status 1
'DynamicJsonBuffer' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
<--- Error log ends

ke5, 2018-04-08:
Apart from the problem described above I should add that using NeoAnimationFX instead of WS2812FX appears to be much more stable. I was experiencing random reboots and wifi hangups with the WS2812FX library using 512 WS2812B leds. It's been running perfectly for the past 4-5 hours so thank you and @debsahu for that.

debsahu, 2018-04-08:
@ke5 Regarding the compile errors when ENABLE_STATE_SAVE_SPIFFS is enabled, install latest [Arduino Json](https://github.com/bblanchon/ArduinoJson)

ke5, 2018-04-08:
Ah! If I enable #define ENABLE_HOMEASSISTANT it compiles ok. I think the problem is that the ArduinoJson library is only loaded if HA is enabled at lines 30-32 of McLighting.ino. Is ENABLE_STATE_SAVE_SPIFFS only needed if HA is enabled? At the moment I don't need HA, MQTT or Button so my thinking is disable them to make the program smaller.

ke5, 2018-04-08:
The Node-Red flow example no longer works properly anymore too - it works once then you have to refresh the page every time so of course you can't access modes, just the colour wheel. 
Should I create a new issue about this? Could it be related to the above issue? 
(It's not a big problem for me, I'm just playing with it at the moment but it is such a nice feature!)

debsahu, 2018-04-08:
Ahh I see, so many permutation and combination of options! I can fix the inclusion of library, create a PR. Regarding node red, start a new issue and tag @toblum as I don't use it.

ke5, 2018-04-08:
Indeed, so many things to play with! Thanks for your help @debsahu 
I'll attempt to solve the Node-Red thing before hassling toblum as I'd like to try to contribute in some small way if I can.

commit 09a26a75 (2018-04-08): fix #125