toblum/McLighting #126: Error compiling with NeoAnimationFX

chkbln, 2018-04-08:
Hello

I get the following error message:
\ libraries \ NeoAnimationFX \ src / NeoAnimationFX.h: 110: 47: error: 'NeoWs2813Method' was not declared in this scope
typedef NeoPixelBrightnessBus <NeoGrbFeature, NeoWs2813Method> NeoPBBGRBws2813;

Best regards

chkbln, 2018-04-08:
An update to the latest NeoPixelBus Versinon was successful. You had the right idea. Thank you.
What is the difference between AMQTT and MQTT? Unfortunately, I have not found a satisfactory answer on the internet. Can you explain that to me in a few short sentences or do you have a link to a comparative presentation?
Best regards

debsahu, 2018-04-08:
Glad that an update fixed the issue. 

AMQTT is a asynchronous protocol, which means it can run in parallel to other things happening like updating leds, wifi etc. This is advantageous over synchronous protocol (MQTT) because in MQTT protocol everything stops and microcontroller checks for new messages. The disadvantage of AMQTT is that one can't do too much computation or can't use delay() and yield() when a MQTT message arrives (MQTT callback function). So what one has to do is just flip some booleans or assign values to variables (non-blocking functions void of delay() and yield()) and perform blocking operations in main loop. When there is no message incoming, there is no overhead on the processor to check for messages in the AMQTT protocol. Hope I was able to explain what I knew!

closed 2018-04-08 by debsahu