toblum/McLighting #128: Feature enhancement for single HTTP call

ryancasler, 2018-04-09:
It would be great if you could send one http call to set brightness, color, speed and mode all in one rather than having to send multiple calls.  If this is already possible and I'm just not seeing it, please let me know.  Otherwise, I think it would make it a lot more user-friendly from an implementation standpoint.

toblum, 2018-04-09:
Hi @ryancasler,

I think you're right. Currently I only see a wayy to set Mode,Color and Speed at once. Not brightness.
e.g. /set_mode?m=3&rgb=FF00AA&s=200

With Websockets or MQTT is shouldn't be a problem. I will see if I can enhance the getArgs() method with brightness. 

Regards
Tobias

ryancasler, 2018-04-09:
You can set the color when you set the mode?  That's not documented in the documentation.  That would be a big help right there.  That would bring me down from 3 http request to 2.  That's a little better.

debsahu, 2018-04-12:
@ryancasler You can now issue 
`/set_mode?m=3&rgb=FF00AA&s=200&c=100` or `/set_mode?m=3&rgb=FF00AA&s=200&p=255`
 Could you check this at my [branch](https://github.com/debsahu/McLighting) before we merge it?

/theaterchase or any "older" named will not accept any argument. Instead use mode "16" `/set_mode?m=16&rgb=FF00AA&s=200&p=255`

bpohvoodoo, 2018-04-12:
Maybe we should remove all those old names.
Or are they still in for compatiblity reasons?

debsahu, 2018-04-12:
Yes, you are right. I think they were carried over from v1 and in here for compatibility reasons.

ryancasler, 2018-04-12:
I can't compile your branch without making a log of fixes/changes.  I've fixd a bunch already but this is the latest:
```
McLighting2:791: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

exit status 1
'class Ticker' has no member named 'active'
```

ryancasler, 2018-04-12:
Brightness doesn't seem to actually change.  Just the speed and mode.  I tried both C and P.  Neither worked.

ryancasler, 2018-04-12:
Also, get color is not returning a correct value.  Sometimes it's 4 characters, sometimes it's 5 sometimes it's 6.  Shouldn't it always be 6 characters?   It seems to be leaving out leading 0's for each pixel.  setting to 010101 returns back 111.

debsahu, 2018-04-12:
`'class Ticker' has no member named 'active'` error is because you need to update to latest arduino-esp8266 (as of today it is v2.4.1). Could you do one more change as well?

In McLighting.ino, change
```
  if (mode == SETSPEED) {
    strip.setSpeed(convertSpeed(ws2812fx_speed));
    mode = HOLD;
  }
```
to
```
  if (mode == SETSPEED) {
    strip.setSpeed(convertSpeed(ws2812fx_speed));
    mode = BRIGHTNESS;
  }
```

ryancasler, 2018-04-12:
2,4.1 has a major memory leak.  It's crashing a whole bunch of other stuff I'm working on.  Had to stick with 2.3.

dcrum, 2018-04-12:
I have this same issue on platformio.  It uses the espressif 1.6 toolchain.

ryancasler, 2018-04-12:
Bingo!!!  That did it.  It can now set the mode, speed, color and brightness all in one.  Nice work!!

debsahu, 2018-04-12:
Ok l'll push that later tonight. 

Regarding memory leak, are you using MQTT? Only MQTT is affected by it. McLighting has AMQTT now as default. And the solution for the leak is to edit the initialization pointer for WiFiClient, which is also posted in the link you posted here.

ryancasler, 2018-04-12:
McLighting isn't but another system I use is VERY affected by it.  So, I haven't been able to update to 2.4.1 yet.  It's either that or keep switching back and forth every 10 minutes.  👎 

Also, if I could be REALLY greedy, it would be oh, so wonderful if "get_status" returned the hex value either in place of or with the RGB value.  Cause now I'm going to have to do a conversion on the upload side to get from RBG back to HEX.

debsahu, 2018-04-12:
Ok regarding rgb conversion,

In McLighting.ino, could you change
```
  server.on("/get_color", []() {
    String rgbcolor = String(main_color.red, HEX) + String(main_color.green, HEX) + String(main_color.blue, HEX);
    server.send(200, "text/plain", rgbcolor );
    DBG_OUTPUT_PORT.print("/get_color: ");
    DBG_OUTPUT_PORT.println(rgbcolor);
  });
```
to
```
  server.on("/get_color", []() {
    char rgbcolor[7];
    sprintf(rgbcolor, "%02X%02X%02X", main_color.red, main_color.green, main_color.blue);
    server.send(200, "text/plain", rgbcolor );
    DBG_OUTPUT_PORT.print("/get_color: ");
    DBG_OUTPUT_PORT.println(rgbcolor);
  });
```

ryancasler, 2018-04-12:
Yup, that worked too.  Very nice!!