toblum/McLighting #131: Could not compile:

schroepfer, 2018-04-11:
Hello i get this error when i want to compile for my NodeMCU: (i changed nothing in source.)

In file included from /Users/Juergen-S/Downloads/McLighting-master/Arduino/McLighting/McLighting.ino:218:0:
sketch/request_handlers.h: In function 'void onMqttDisconnect(AsyncMqttClientDisconnectReason)':
request_handlers.h:1132: error: 'TLS_BAD_FINGERPRINT' is not a member of 'AsyncMqttClientDisconnectReason'
       if (reason == AsyncMqttClientDisconnectReason::TLS_BAD_FINGERPRINT) {
                     ^
request_handlers.h:1146: error: 'ESP8266_NOT_ENOUGH_SPACE' is not a member of 'AsyncMqttClientDisconnectReason'
       } else if (reason == AsyncMqttClientDisconnectReason::ESP8266_NOT_ENOUGH_SPACE) {
                            ^

debsahu, 2018-04-11:
AMQTT of enabled by default, so have you installed async mqtt libraries?

See wiki for other requirements:

AMQTT integration (enabled by default)

Async MQTT by @marvinroger (tested with 0.8.1)
Only when you have activated AMQTT in definitions.h.
https://github.com/marvinroger/async-mqtt-client
Needs ESPAsyncTCP: https://github.com/me-no-dev/ESPAsyncTCP
(Download the library repository as ZIP. Load that library into to Arduino software by: "Sketch > Include Library > Add .ZIP Library".)

schroepfer, 2018-04-11:
Ohoh... now it works!
Thank you

closed 2018-04-11 by debsahu