toblum/McLighting #133: Modes Page Reverts Color

ryancasler, 2018-04-13:
When I open the second pane on the webpage for mode selection, it always reverts back to RGB 128,128,128.  The brightness and speed sliders don't match what is currently running on the strip but they don't "hijack" what is already running at least.  But, for example, I initiate a group of settings from another source.  Now i want to go in and dim them more, i can't do that since the slider is only on the 2nd page and I'll lose my color setting.  Any assistance you could give me would be greatly appreciated.

ryancasler, 2018-04-25:
I see a lot of movement on the project.  Anything on this?

toblum, 2018-04-25:
Not yet, creating a complete new UI takes time (I also have to learn a new framework). Expect more weeks than days for the first results.

toblum, 2018-05-20:
@ryancasler PLease have a look here: https://github.com/toblum/McLightingUI/

Regards
Tobias

closed 2018-05-20 by toblum

ryancasler, 2018-05-21:
Will do when I'm back home this weekend.

ryancasler, 2018-05-25:
Very confused.  All I have to do is upload the index2.html file?  And then what?  Do I delete the old one?

debsahu, 2018-05-25:
You can enter http://<IP>/index2.html and check if that version works. I can confirm it does! If you are satisfied, delete both files on the SPIFFs and rename index2.html on your computer to index.html and upload it.

chkbln, 2018-05-27:
Hi debsahu,
can you tell me how to delete files in the SPIFFs?
Thanks and best regards

debsahu, 2018-05-27:
@chkbln Using Arduino IDE, goto Tools -> Erase Flash -> All flash contents and flash McLighting.

LeonVos, 2018-05-27:
In sketch you can do;
SPIFFS.format();

chkbln, 2018-05-27:
I was hoping to delete files via Edit

[image]