toblum/McLighting #135: Cant connect to the node after setting up Wifi

rklueber, 2018-04-14:
Hello,

Cool project. Thank you very much. I am stuck. I tried everything I could imagine. I am at the point where I need some help please.

I did
* checked I have all libraries in the versions mentioned in the [Software Section](https://github.com/toblum/McLighting/wiki/Software-installation)  
* compile and uploaded the sketch
* uploaded the files from /data
* connected to the the serial monitor
* connected to the wifi of the ESP and got to 192.168.4.1
* was able to set SSID and password
* see the ESP connecting to my wifi getting an IP address vie the serial monitor
* in no way I can connect to the ESP which is not connected to my wifi
* I tried with this repro and with the FabLab-Luenen one. Same thing.

I am really looking forward to get a hint what could been wrong. I tried even on two different ESP8266 hardware with the same result.

Thanks in advance.

Regards
Ralf

`

*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
Entered config mode
192.168.4.1
McLighting01
*WM: 
*WM: Configuring access point... 
*WM: McLighting01
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
connected...yeey :)
Open http://192.168.30.79/ to open McLighting.
Use http://McLighting01.local/ when you have Bonjour installed.
New users: Open http://192.168.30.79/upload to upload the webpages first.

Coudnt find "/stripstate.json"
 Failure!
*WM: freeing allocated params!
`

rklueber, 2018-04-15:
Not via the URL, I cant reach the URL. I did vie the plugin
`
[SPIFFS] data   : /xxx/Documents/Arduino/McLighting/data
[SPIFFS] size   : 1004
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/edit.htm.gz
/favicon.ico
/graphs.js.gz
/index.htm
skipping .DS_Store
[SPIFFS] upload : /var/folders/qm/wv4yj_ds2wn5fzxhdvj3p6600000gn/T/arduino_build_544920/McLighting.spiffs.bin
[SPIFFS] address: 0x300000
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : /dev/cu.SLAB_USBtoUART
[SPIFFS] speed  : 115200

warning: espcomm_send_command: wrong direction/command: 0x01 0x00, expected 0x01 0x08
Uploading 1028096 bytes from /var/folders/qm/wv4yj_ds2wn5fzxhdvj3p6600000gn/T/arduino_build_544920/McLighting.spiffs.bin to flash at 0x00300000
................................................................................ [  7% ]
.............................`

rklueber, 2018-04-15:
Hi Tobias,

One board I checked is this:
https://learn.adafruit.com/adafruit-feather-huzzah-esp8266?view=all

Another one was wemos.cc LoLin, new NodeMcu V3

Both, same issue

Board: NodeMCU 1.0 (ESP-12E Module)
CPU Frequency: 80MHz, tried 160 as well
Flash Size: 4M(1M SPIFFS)
IwIP Variant: v2 Higher Bandwidth

Is there a way to check wether the webserver is running? A Portscan on 192.168.30.79 gives nothing.

I connected WS8211 LEDs and a button. The LEDs and button work as expected.

Regards
Ralf

rklueber, 2018-04-15:
Hi,

I was able to get all working. 

I have set up a totally ne wifi hotspot. In this environment I could get to the webserver. I dont know why it did not connect to my normal wifi. Anyhow. Seems nothing related to the code of this project.

Apologies for any inconvenience caused.

Works as a charm. Thank you.

Regards
Ralf

closed 2018-04-15 by debsahu

toblum, 2018-04-15:
Hi @rklueber,

no problem. Great that it works now.

Regards
Tobias