toblum/McLighting #137: Keyboard shortcuts for McLighting (enhancemnet)

flagstone78, 2018-04-15:
This project is really awesome! It really helped me get my lights up and running quickly.

I have my ws2812b leds around my computer area, so i wanted a way to control the lights directly from my keyboard without having to wait for a web page to load every time i wanted to adjust them.

So, I made an AutoHotKey script that is able to talk directly to the webserver!
I made this on a windows 7 machine. I dont know what the traytip will look like on other operating systems. On windows7 it makes a little balloon popup in the bottom corner that automatically disappears after a few seconds. The shortcuts use the numberpad. Make sure numlock is on

The shortcuts are as follows: 
ctrl+windows+numpad4 decrements the animation mode
ctrl+windows+numpad6 increments the animation mode

ctrl+windows+numpad5 turns off the lights
ctrl+windows+numpad0 turns on all of the lights to the current color

ctrl+windows+numpadPlus increases the brightness
ctrl+windows+numpadSubtract decreases the brightness

ctrl+windows+numpadMultiply increases the speed
ctrl+windows+numpadDivide decreases the speed

ctrl+windows+numpad7 increments the red component of the current color
ctrl+windows+numpad1 decrements the red component of the current color

ctrl+windows+numpad8 increments the green component of the current color
ctrl+windows+numpad2 decrements the green component of the current color

ctrl+windows+numpad9 increments the blue component of the current color
ctrl+windows+numpad3 decrements  the blue component of the current color

If people like this, I could try to have it talk to the mqtt server so that it stays up to date with the current mode, color, speed, and brightness.

The AutoHotKey script: (make sure to replace the ip with the ip of your controller)
```
#NoEnv
#SingleInstance force ; only lets one run
SetBatchLines, -1

socket := new Example("ws://192.168.1.33:81") ; replace with the ip address of the mclighting controller

incrementAmount := 10 ; how finely you adjust the color, speed, and brightness per keypress
maxColor := 255
maxMode := 56 ; the maximum mode id number

red := 127
green := 127
blue := 127

brightness := 127
speed := 127
mode := 1

^#Numpad4:: ; switch modes down
	mode := mode - 1
	if(mode < 0)
		mode := maxMode
	cmd = /%mode% ; / is the set effect mode command
	TrayTip, , Mode: %mode%
	socket.Send(cmd)
	; MsgBox %cmd%
	return

^#Numpad6:: ; switch modes up
	mode := mode + 1
	if(mode > maxMode)
		mode := 0
	TrayTip, , Mode: %mode%
	cmd = /%mode%
	socket.Send(cmd)
	; MsgBox %cmd%
	return
	
^#Numpad5:: ; turn off
	TrayTip, , Leds are off
	socket.Send("=off") ; = is the set control command
	return
	
^#Numpad0:: ; turn on to static color
	TrayTip, , Leds are on
	cmd := "*"+toHexColor(red,green,blue) ; * is the set all command
	socket.Send(cmd) ; = is the set control command
	return
	
	
^#NumpadSub:: ; decreases brightness
	brightness := brightness - incrementAmount
	if(brightness < 0)
		brightness := 0
	cmd = `%%brightness% ;  % is the set brightness mode command. ` is the escape character
	TrayTip, , Brightness: %brightness%
	socket.Send(cmd)
	; MsgBox %cmd%
	return
	
^#NumpadAdd:: ; increases brightness
	brightness := brightness + incrementAmount
	if(brightness > maxColor)
		brightness := maxColor
	cmd = `%%brightness% ; % is the set brightness mode command
	TrayTip, , Brightness: %brightness%
	socket.Send(cmd)
	; MsgBox %cmd%
	return

^#NumpadDiv:: ; decreases speed
	speed := speed - incrementAmount
	if(speed < 0)
		speed := 0
	cmd = ?%speed% ;  ? is the set brightness mode command
	TrayTip, , Speed: %speed%
	socket.Send(cmd)
	; MsgBox %cmd%
	return
	
^#NumpadMult:: ; increases speed
	speed := speed + incrementAmount
	if(speed > maxColor)
		speed := maxColor
	cmd = ?%speed% ;  ? is the set brightness mode command
	TrayTip, , Speed: %speed%
	socket.Send(cmd)
	; MsgBox %cmd%
	return

^#Numpad7:: ; increase red amount
	red := red + incrementAmount
	if(red > maxColor)
		red := maxColor
	
	cmd := "#"+toHexColor(red,green,blue) ; # is the set main color command
	; MsgBox %cmd%
	TrayTip, , Red: %red%
	socket.Send(cmd)

	return
	
^#Numpad1:: ; decrease red amount
	red := red - incrementAmount
	if(red < 0)
		red := 0
	
	cmd := "#"+toHexColor(red,green,blue) ; # is the set main color command
	; MsgBox %cmd%
	TrayTip, , Red: %red%
	socket.Send(cmd)

	return

^#Numpad8:: ; increase green amount
	green := green + incrementAmount
	if(green > maxColor)
		green := maxColor
	
	cmd := "#"+toHexColor(red,green,blue) ; # is the set main color command
	; MsgBox %cmd%
	TrayTip, , Green: %green%
	socket.Send(cmd)

	return
	
^#Numpad2:: ; decrease green amount
	green := green - incrementAmount
	if(green < 0)
		green := 0
	
	cmd := "#"+toHexColor(red,green,blue) ; # is the set main color command
	; MsgBox %cmd%
	TrayTip, , Green: %green%
	socket.Send(cmd)

	return
	
^#Numpad9:: ; increase blue amount
	blue := blue + incrementAmount
	if(blue > maxColor)
		blue := maxColor
	
	cmd := "#"+toHexColor(red,green,blue) ; # is the set main color command
	; MsgBox %cmd%
	TrayTip, , Blue: %blue%
	socket.Send(cmd)

	return
	
^#Numpad3:: ; decrease blue amount
	blue := blue - incrementAmount
	if(blue < 0)
		blue := 0
	
	cmd := "#"+toHexColor(red,green,blue) ; # is the set main color command
	; MsgBox %cmd%
	TrayTip, , Blue: %blue%
	socket.Send(cmd)

	return
	
toHexColor(r, g, b)
{
	SetFormat, IntegerFast, hex ; To print values as hexadecimal
	colorNum := r*65536 + g*256 + b
	color = 000000%colorNum% ; convert to string and add preceeding zeros
	color := StrReplace(color, "0x") ; replace the 0x from the middle
	color := SubStr( color, -5 ) ; get last 6 characters
	
	SetFormat, IntegerFast, dec ; sets the print mode back to decimal
	return ""+color ; 
}

class Example extends WebSocket
{
	OnOpen(Event)
	{
		TrayTip, ,Connection established to led controller!
		; InputBox, Data, WebSocket, Enter some text to send through the websocket.
		; this.Send(Data)
	}
	
	OnMessage(Event)
	{
		; MsgBox, % "Received Data: " Event.data
		; this.Close()
	}
	
	OnClose(Event)
	{
		TrayTip, , Websocket Closed
		this.Disconnect()
	}
	
	OnError(Event)
	{
		TrayTip, , Websocket Error
	}
	
	__Delete()
	{
		TrayTip, , Exiting
		ExitApp
	}
}

class WebSocket
{
	__New(WS_URL)
	{
		static wb
		
		; Create an IE instance
		Gui, +hWndhOld
		Gui, New, +hWndhWnd
		this.hWnd := hWnd
		Gui, Add, ActiveX, vWB, Shell.Explorer
		Gui, %hOld%: Default
		
		; Write an appropriate document
		WB.Navigate("about:<!DOCTYPE html><meta http-equiv='X-UA-Compatible'"
		. "content='IE=edge'><body></body>")
		while (WB.ReadyState < 4)
			sleep, 50
		this.document := WB.document
		
		; Add our handlers to the JavaScript namespace
		this.document.parentWindow.ahk_savews := this._SaveWS.Bind(this)
		this.document.parentWindow.ahk_event := this._Event.Bind(this)
		this.document.parentWindow.ahk_ws_url := WS_URL
		
		; Add some JavaScript to the page to open a socket
		Script := this.document.createElement("script")
		Script.text := "ws = new WebSocket(ahk_ws_url);`n"
		. "ws.onopen = function(event){ ahk_event('Open', event); };`n"
		. "ws.onclose = function(event){ ahk_event('Close', event); };`n"
		. "ws.onerror = function(event){ ahk_event('Error', event); };`n"
		. "ws.onmessage = function(event){ ahk_event('Message', event); };"
		this.document.body.appendChild(Script)
	}
	
	; Called by the JS in response to WS events
	_Event(EventName, Event)
	{
		this["On" EventName](Event)
	}
	
	; Sends data through the WebSocket
	Send(Data)
	{
		; MsgBox %Data%
		this.document.parentWindow.ws.send(Data)
	}
	
	; Closes the WebSocket connection
	Close(Code:=1000, Reason:="")
	{
		this.document.parentWindow.ws.close(Code, Reason)
	}
	
	; Closes and deletes the WebSocket, removing
	; references so the class can be garbage collected
	Disconnect()
	{
		if this.hWnd
		{
			this.Close()
			Gui, % this.hWnd ": Destroy"
			this.hWnd := False
		}
	}
}
```
Credit goes to [G33kDude/WebSocket.ahk](https://github.com/G33kDude/WebSocket.ahk) for the websocket class

debsahu, 2018-04-15:
This is fantastic! Love this approach, I will surely try this out.

toblum, 2018-04-15:
Hi @flagstone78,

I haven't used AutoHotKey so far, but will surely give it a try. This is also neat for testing.

If you want, we can put that script into the clients directory of McLighting.

Best regards
Tobias

flagstone78, 2018-04-15:
@toblum I made the pull request to make an AutoHotKey directory in the clients folder.  It also contains an updated version of the script that handles server timeouts and errors better.

closed 2018-04-17 by flagstone78