toblum/McLighting #139: Help wanted | Will it work?

gabenthegamer, 2018-04-16:
Hello :D I have the chip as in the picture it works i see web panel but i dont have my led strips yet. But when i get them do i solder data to pin 1 and + and minus to gnd and vin? 

https://i.imgur.com/clx5Ifm.jpg

Also here is my led strips!

gabenthegamer, 2018-04-16:
And if strip wont work can somone find a strip that works? Thanks!

toblum, 2018-04-16:
Hi @gabenthegamer,

the chip is ok. Wiring also. Have a look here for more information on powering the strip: https://learn.adafruit.com/adafruit-neopixel-uberguide/basic-connections

I’m not sure about the strip. SK6812 are usually RGBW strips. Yours is rated as RGB. Nevertheless it should work with the library. You maybe have to select a different mode for the strip if colors are swapped. Just ask if that’s the case. 

Normally all WS 2811 and WS2812 type strips work without changing anything. 

Regards 
Tobias

closed 2018-04-16 by toblum

gabenthegamer, 2018-04-18:
Some more questions. Will this work? 
https://banggood.app.link/yPmRiiZNdM
What do I need to solder? And if I need external power how do I do that? Just link everything I need😀 Preferably on banggood

debsahu, 2018-04-18:
yes but use ws2812b strips (runs on 5V). Not the cheapest on banggood, link: https://www.banggood.com/5M-90W-300SMD-WS2812B-LED-RGB-Colorful-Strip-Light-Waterproof-IP65-WhiteBlack-PCB-DC5V-p-1035641.html?rmmds=search&cur_warehouse=CN

gabenthegamer, 2018-04-18:
How do I power it? Just via the esp?

debsahu, 2018-04-18:
Use at least a 5V 10A power supply for 5m/300LED setup. You can power the esp8266's Vin using the same power supply as well. Remember to adhere to guidelines posted here https://learn.adafruit.com/adafruit-neopixel-uberguide/basic-connections

gabenthegamer, 2018-04-18:
I removed all the pins on the esp is that a problem? Just to make it more compact

gabenthegamer, 2018-04-18:
Will this work? https://www.banggood.com/5M-45W-150SMD-WS2812B-LED-RGB-Colorful-Strip-Light-Waterproof-IP65-WhiteBlack-PCB-DC5V-p-1035640.html?rmmds=search&cur_warehouse=CN Its less watt idk what that does XD

debsahu, 2018-04-18:
I don't see why not, the LEDs are ws2812b. This has half the number of LEDs 150 (30/m), so power requirement is half!

gabenthegamer, 2018-04-18:
But is there a way to make like a datasheet of all the parts I need to make it? From power to esp?

gabenthegamer, 2018-04-18:
This is my scematic is that correct? Or where does the powersuply go in?

https://i.gyazo.com/4464ed339caee2ed943b95d5a381148f.png

debsahu, 2018-04-18:
Everything I mention below is described in detail here, please read: https://learn.adafruit.com/adafruit-neopixel-uberguide/basic-connections
* Add a 1000uF capacitor on your 5V input
* Use a 3.3V to 5V level shifter (a chip like: 74AHCT125) between data pin on nodemcu and data_in. Also remember to add <500ohm resistor between level shifter and strip's data_in.
* Dont power 5V >1A via USB of NodeMCU, instead connect to Vin like you do in your schematic.
* Last thing is I would use D4 instead of D1, just in case you want to try NeoAnimationFX (use UART method).
* Use thick good quality wires to carry >1A current, especially the power connections.

gabenthegamer, 2018-04-18:
The 1000uf on banggood says 35v?

debsahu, 2018-04-18:
It will work, 35V is the max voltage that the capacitor will withstand before it explodes! Since 5V < 35V you are good.

gabenthegamer, 2018-04-18:
Is there a way you can send all the parts with banggood links I need? Thanks slot😀

gabenthegamer, 2018-04-18:
Alot*