toblum/McLighting #140: AP Mode

PhynixP, 2018-04-18:
Hello, it is possible to use this Software
without a Router.
For example on a place without Lan (outdoor)

debsahu, 2018-04-18:
See https://github.com/toblum/McLighting/issues/29 Not possible for now. Lot of things are loaded from the internet on the webpage that can't be served on the esp8266. A major code change is required to accommodate standalone AP mode.

TechOnlyGamer, 2018-04-18:
but how about using ur phone as mobile hotspot? set the esp up to run wlan from there then using a diy app for controls using hhtp rest e. g.? worked on an app a few days ago with app inventor but stopped to see where ur restAPI changes end.

debsahu, 2018-04-18:
@TechOnlyGamer now the restapi can handle control of strip (mode, color, brightness, speed) using just one call "/set_mode" see https://github.com/toblum/McLighting/issues/128

PhynixP, 2018-04-18:
Thanks for this great idea with the telefon hotspot ☺

jmyrik28, 2018-05-21:
ws2812fx.setSegment(i, seg["start"], seg["stop"], seg["mode"], _colors, seg["speed"], reverse ? REVERSE : NO_OPTIONS);

debsahu, 2018-05-21:
@jmyrik28 update to the latest WS2812FX.

jmyrik28, 2018-05-21:
@debsahu  LOL 
Archiving built core (caching) in: C:\Users\MI\AppData\Local\Temp\arduino_cache_133232\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_30e578dae652c3bcab6c3619d8340955.a
sketch\sketch_may22a.ino.cpp.o:(.text._Z10handleRootv+0x8): undefined reference to `index_html'

collect2.exe: error: ld returned 1 exit status

exit status 1

debsahu, 2018-05-21:
@jmyrik28 REVERSE was introduced in the latest WS2812FX, that's why I asked you to update. `extern const char index_html[];` references to a cpp file containing the index.html. And I'm guessing it is not McLighting interface. @BloodADM could elaborate on how to get access to it, but what I like on the code posted is that there is a **template for using segments**. BTW check https://github.com/toblum/McLightingUI for complete standalone web interface developed by @toblum.

BloodADM, 2018-05-22:
Ive moddified my code for more stable version of AP last one was hanging, and some times stoping AP mode, now it works like charm :) another thing to do is to make it work on every browser becouse right now it works only on chrome :P, the effects are on https://www.instagram.com/anwebpl/ :)

BloodADM, 2018-05-29:
And fixed another issue, after few restarts of Nodemcu the website didn't loaded properly, the fix was to insert js scripts and css into index.html.cpp as <script></script> and <style></style> now the code is P&P :)

toblum, 2018-12-16:
Closing issue due to inactivity.

closed 2018-12-16 by toblum