toblum/McLighting #142: Android App for controlling McLightning

TechOnlyGamer, 2018-04-19:
Hey Guys,
first of all thank you all very much for making this project better and better.
As i posted in another Isuue, i started creating an App for myself using MIT AppInventor 2(cause iam a noob at programming Java), so that i dont have to set up my Light via Browser or Homeassistant everytime.

After your last update with the single http request i could finished it in very early Alpha.
It is a hobby i like to do in my free time, iam very new in programming and have a lot more to do to make the App better but its a start. If you like it, maybe someone could help with Ideas, Graphics to make it a lot cooler etc. 

[McLightningApp.zip](https://github.com/toblum/McLighting/files/1928816/McLightningApp.zip)

debsahu, 2018-04-20:
Wow, works!

kokopaulc, 2018-04-20:
Awesome !!!👌🤘🤘🤘

toblum, 2018-04-20:
Hi @TechOnlyGamer,

I was able to try it out today. Worked great on my spare Android phone. If you have the app published somewhere, I'll be delighted to link to the download, if you like.

Best regards
Tobias

closed 2018-04-20 by toblum

TechOnlyGamer, 2018-04-20:
i checked the requirements to publish the app in the playstore to get better access for all finding this project but it cost me an initial amoint of 25 usd. so o think the way giving u the files for the rep is better for me and cheaper

toblum, 2018-04-20:
OK, great!
I put the .apk to the folder https://github.com/toblum/McLighting/tree/master/clients/Android
Let me know if you want to add something or maybe have a readme file with more informations.

Thank you for your contribution!

TechOnlyGamer, 2018-04-20:
u can tell me how i can edit things or upload new versions for u to ckeck and merge i dont use github often other for reading xD

toblum, 2018-04-20:
To add new versions or files you need to create a push request via git. This needs some experience with git, but you'll find easy explanations how to so that in the web. Or just contact me and I will update it, no problem.
If I'm right AppInventor also has a kind of gallery where you can publish apps, but I never used it.

debsahu, 2018-04-20:
Is there a button for turning off the lights, it was one thing I was searching for.

TechOnlyGamer, 2018-04-20:
not right now... one of my things on my todo list but its a short implementation i work in shortly. i even thought about a config option for the wifimanager part but i need to work out the api from the project maybe u could help me there if i reach that stage?

debsahu, 2018-04-20:
Sure thing

ryancasler, 2018-04-21:
Hey @TechOnlyGamer ...this is really amazing work. And you say you're not even a programmer?  I looked into MIT App Inventor and it looks really amazing.  In order to work on this, you'd need to share the aia file though.  Anyway you could attach that as well?  I'd love to tinker around for a bit.

Another thought, if you're looking for features to add, would be if in the settings screen you could enter multiple IP addresses and then have a checkbox next to each.  That way you could choose which controller or combination of controllers you're changing within the app.
Of course, I have absolutely no idea how to do that...but it sounded cool so thought I'd pass the idea on.  

But this is REALLY amazing work.

ryancasler, 2018-04-21:
Thanks a bunch!!

BezziHD, 2018-05-03:
Hey @TechOnlyGamer 

Can you try to add some more IP_Adresses cause i want to control three ESP8266 with youre App.
Or do you know how I can solve this Problem?

Thanks for your App works pretty good :)

TechOnlyGamer, 2018-05-03:
sure i will! i already have a few ideas how to solve it and try to get it to work at the weekend i think!

BezziHD, 2018-05-03:
Thank you so much!
It would be nice if you could notify me when your new Version is ready :)

And another thing:
Is it possible to start the App without resetting the Animation > It always turns white when I connect my Smartphone with it :D 
It isnt to bad cause you know that it is connected xD But I think you know what i mean

Thanks :)

TechOnlyGamer, 2018-05-03:
sure didnt know that this is even a thing. ill post an update in this thread

debsahu, 2018-05-11:
@TechOnlyGamer His difficult is it to listen on a udp port for a multicast/unicast message on a Android app?

TechOnlyGamer, 2018-05-11:
for me yes. with app inventor for sure and i dont even know how to handle this things cause it is not qrotten in java i have to work with what app inventor can offer