toblum/McLighting #145: Must Enable Legacy Animations for Off Button to Work

ryancasler, 2018-04-22:
If you don't enable Legacy Animations, the off button on website doesn't work to turn off the LEDs.  Also, the static function and I believe the breath do not work as well.  Enabling legacy animations resolved the issue.

debsahu, 2018-04-22:
The off button calls "=off". That is a legacy mode, same with TV ("=tv"). The web front-end needs modification to call <HOSTIP>/off instead, for off button action. For now don't disable legacy mode, it is a work in progress. I put it in there so that later on if we decide to retire those modes all together it would be easier. Regarding static and breath modes, they are from ws2812fx and the same errors should occur even weather legacy mode is enabled or disabled, I will check this later and keep this thread updated on that front.

debsahu, 2018-04-22:
I disabled legacy mode and tested, both static and breath work when called from websockets. Any more details on how to replicate the second part of this issue?

@toblum Whenever you have time, could you disable legacy mode and check how the web-interface responds to "OFF" and "TV" button. Can we disable them on the web-interface if it is not present in "mode"? I dont know nothing about designing/editing web-interface.

ryancasler, 2018-04-22:
I would say you can't really remove the off button.  That's kinda of important.

ryancasler, 2018-04-23:
Static and Breath work.  The issue was that after the off button is pushed nothing works.  I have enabled legacy automations and it still doesn't work.  So the only way to turn them off is to manually enter it in the URL.  This isn't really workaable long term.  Either the legacy automations should be removed until the html interface can be redesigned or the web interface should be fixed.

debsahu, 2018-04-23:
@toblum Sounds interesting, I think adding legacy modes (if enabled) to listModesJSON() is the best way to go.

toblum, 2018-04-24:
@debsahu OK, I think we can keep this issue open and I will have a look at it, when I'm back at my workbench.