toblum/McLighting #146: HTTP Rest API No Longer Returns Command Set by Website

ryancasler, 2018-04-22:
After setting a mode by http GET request, it is displayed corrected in the status GET.  However, if you then go in and make changes via the website to another mode or brightness or color, those changes are not reflected the next time you perform an GET status request.  The info returned is what was set by the last GET request.  It appears that mode is not being stored.  I've already updated all of my sketches with the new code.  Any way we can pull those last commits?

debsahu, 2018-04-22:
@ryancasler When you load the webpage, did you start a new session? If you have the webpage open and upload new code from IDE/OTA, websocket connection gets disconnected. When you make changes to the modes on the web interface do the LEDs respond accordingly?

ryancasler, 2018-04-22:
Yes, the LEDs respond correctly.

Ryan Casler
617.504.9597 (mobile)
ryancasler@gmail.com

debsahu, 2018-04-22:
@ryancasler good that you removed your personal info, I was going to edit it. Regarding this issue, I was able to replicate it and am pushing an update. Could you try testing from my branch?

ryancasler, 2018-04-22:
I think so...the only changes you made were to the request_handlers.h file correct?  I'm actually working on one right now.  So I will update and give it a shot.

debsahu, 2018-04-22:
@ryancasler yes, https://github.com/toblum/McLighting/pull/148/files

ryancasler, 2018-04-22:
Yup...that worked.  When I make a change on the http frontend I'm able to see that accurately reflected back to me form the GET message.

closed 2018-04-22 by debsahu