toblum/McLighting #147: McLighting Mesh ??

LeonVos, 2018-04-22:
Hi Tobias, Reading about the Android app :-) Is it an (great) idea to implement a mesh for all Mclithing esp's ?? ESP8266WiFiMesh or Wifiudpclient/server (#ifdef) or even BROADCAST?
Not sure yet if there need to be a mas of code to be changed (like Server.on /....... etc)

But know for sure many would like to control ALL there McLighting esp's without a Mosquito/mqtt server but from the buildin webpage.

to be cont.....................

Regards Leon

TechOnlyGamer, 2018-04-22:
i think if u want to control all of them from the app i dont need to change anything on the esp. iam planning right know to implement more then one device in my app so that i could even call the function to post the http request to all of them. a mesh is more a way i think that all devices communicate with each other but the app can handlr more devices and chamge all of them so the work is done by tje app not the esp so we could keep the code on tje esp smaller and more efficient

LeonVos, 2018-04-22:
When one has a Android phone, yes,  but for iphone there's no app.
LV

TechOnlyGamer, 2018-04-23:
yeah but its the same thing, i think there is too much work for the esp to get this working the approach with an rpi running only mqtt is like 20 bucks??? and time beeing maybe we goet someone taking the same approach in creating an iphone app or something.

debsahu, 2018-04-23:
A big **TODO**, not an easy implmentation! For now the recommendation is to use MQTT and all clients connect to same topic. Alternatively as mentioned here call multiple rest API from one place (like an app). I've no app building experience.

I think WiFiMesh is a great solution for adhoc wifi network. When I have time I will try to get some examples of painlessMesh working. https://gitlab.com/painlessMesh/painlessMesh/blob/master/examples/webServer/webServer.ino 
* Never done any mesh networking ever, so ideas welcome here! I will create a separate branch (or even a separate project) and see how this progresses. 
* As mentioned in https://github.com/toblum/McLighting/issues/140 it will require having a stand-alone web interface (I've no experience in building one), or if others have ideas on how to connect just one node to WiFi and rest to on the mesh.

toblum, 2018-04-23:
Hi @LeonVos,

using a mesh network sounds interesting but is not easy to implement in such a way that it works flawlessly for all users. I'd like to have it as easy and stable as possible. 
@TechOnlyGamer approach to send the same command to a number of ESPs looks like a easier solution. 
I also started working on a alternative web UI that could talk to more than one ESP, but it will take some time to find a nice and stable solution.

Best regards
Tobias

debsahu, 2018-04-24:
Update: I have something working as of now,
* Uses painlessMesh to create mesh network and broadcasts state to every node
* Connects to known wifi network predefined in code (no wifimanager, have to create custom webpage for this)
* Web interface is borrowed from WS2812FX esp8266 example, completely served on ESP8266
* Can do minimal file upload to SPIFFs
* Completely Async!
* Uses Task Scheduler, no more tickers
* No delays in entire code
* Has RESTful API (Async WebServer)
* Async Websockets (working on it)
* Async MQTT (not implemented as of now)

@toblum It does not have the McLighing interface that you have here, but do you want me to create a branch for people to try?

TechOnlyGamer, 2018-04-24:
if there is an RestAPI say Hello to my Android App

toblum, 2018-04-25:
Hi @debsahu,

that sounds great. Can't wait to try that out. Will see if I can make the webinterface work for that. Is the API mainly the same?

Sure, create a branch as you like.

Regards
Tobias

debsahu, 2018-04-25:
@toblum Async Websockets and Async MQTT API are the same. RESTful API is working and is same as ws282fx example for now. How do you test WebSockets, i.e. your favorite client?

Currently working on WiFiManager to set WiFi for internet, for now it is hard-coded. Also exporting tickers to Task Scheduler. Work in progress!

debsahu, 2018-04-25:
@toblum I created develop and experimental. Since this is experimental I will add my code there, not ready for public consumption yet. I cant delete any of your branches, I think only you can do that by going to 'settings'. I also recall that one can restore deleted branches, if needed as well.

toblum, 2018-04-25:
Hi @debsahu,

I deleted some branches now. Much better overview. I hope I didn't delete too much.

I was using this chrome extension: https://chrome.google.com/webstore/detail/simple-websocket-client/pfdhoblngboilpfeibdedpjgfnlcodoo It has all the basic features and "just works".

Best regards
Tobias

debsahu, 2018-04-30:
Anyone here willing to try the Meshy McLighting? https://github.com/toblum/McLighting/tree/experimental/Arduino/MeshyMcLighting

**Meshy McLighting Server**
WiFi and MQTT settings are still hard coded (Any ideas of making use of something like WiFiManager or something you have come across is appreciated). WiFi channel of the router needs to be manually specified, usually its channel 6 in USA as default.

**Meshy McLighting Clients**
Do not enable WEBSERVER for now, not being used. WiFi channel must be the same as the WiFi router and Server.

@toblum created a favicon.ico, I dont know if you like it? It is there for me to see if an icon is displayed on chrome, we can change it because I spent a whole 3 min to creating it! And what I need help with is it correctly implemented in the webpages `<link rel='shortcut icon' type='image/x-icon' href='favicon.ico' />`? I dont see the icon thats why I'm asking.

toblum, 2018-05-03:
Hi @debsahu,

I hope I'll find time at the weekend to try it out. So little time at the moment. I'm also working on an alternative UI for McLighting (https://github.com/toblum/McLightingUI). I'm trying to have a first version ready soon.

MeshyMcLighting seems to have a similar API, so it should be possible to build a UI that works on both. Did I see this right? So both together could be the new and improved v3 of McLighting. What do you think?

Regarding Favicon: I like it. But if we choose an icon we should have also bigger versions of it, so that we can provide homescreen icons for mobile devices. I havn't done that so far, because I didn't find a cool icon. Maybe some of the users has also some skill to create one. 

All I can say again is that I'm totally impressed with your work. You can do in a short time what would have cost me months and years. Thanks again for your contribution.

Best regards
Tobias

toblum, 2018-05-16:
Hi Deb,

this looks great. I like how reliable the sync happens. This is a much better approach than my experiments with websocket calls to multiple nodes. This isn't really reliable as the ESP struggles with many WS connections open.
I see that the master node can also work standalone. I first thought that you always need at least two nodes, but this is great. So this version could easily replace the current implementation.
Regarding power consuption: I don't think it's a problem in most cases, as the strips draw the most power. But is it possible to turn off mesh functionality if you don't need it to save power? I didn't see that.
As the API is almost similar, I will try to make the new UI work with your implementation. Should be easy.
I had to order fresh ESPs and strips to have a meshy setup in parallel to my existing one. I hope they will arrive soon.

Greetings
Tobias

debsahu, 2018-05-16:
@toblum A few things to note,

1. WS2812FX + painlessMesh + TaskScheduler causes crashes. Even with setting no delays on loop there is some interference of SDK call [espShow()](https://github.com/adafruit/Adafruit_NeoPixel/blob/4d0987ede8f8c204ff333de9a2fcbb70c646b7a5/Adafruit_NeoPixel.cpp#L115) in NeoPixels library. NeoAnimationFX + painlessMesh + TaskScheduler has no issues.
2. From my numerous experiments, IwIP 1.4 higher bandwidth (default settings on platformio, not default on ArduinoIDE) is reliable and IwIP 2.0 is a mess.
3. For using McLightingUI (also in the video), ws calls are exactly the same so I replaced "**:81**" with "**/ws**", very minimal code change.
4. I have a Server without Mesh (uses native TaskScheduler and not painlessMesh's stripped down version) working non-stop (300LEDs), still testing on WS2812FX and NeoAnimationFX settings. I am having trouble with mDNS and DNS for some reason. I will iron this out when I have time and create a `#define DISABLE_MESH_SETTINGS` that can be toggled in `definitions.h`. Or I can upload what I got and someone can pitch in with creating this definition.

LeonVos, 2018-05-17:
Jep, looks like the mesh was a good brainer of me , @debsahu GOOD JOB.
@toblum, nice job on the new controls.

And some new thing about the wifimanager, checkout ;
https://github.com/kentaylor/WiFiManager#portal-continues-to-exist-after-configuration
Ken has leveled this up, now when on dhcp you get feedback from the wifimanager on connection status and addresses, great idea.

debsahu, 2018-05-19:
@toblum use IwIPv1.4 Higher-Bandwidth

debsahu, 2018-05-19:
Also what is the serial output on the client side? It should say connected to AP. On the server Serial output, it should say the WiFi channel that the clients needs compiling, does it match the WiFi channel on the router? I see that in your output you posted it says it's connected to 1 node, so I'm guessing client is connected? Could you change color or something ok the server to trigger a broadcast and see if syncing with client happens?

toblum, 2018-05-19:
I had already selected this. But after reflashing multiple times it works now. I don't know what was wrong.

I have now one server with two clients. They all sync, but it takes always about 5 seconds until a change is reflected at the clients. In your video it was a instant reaction. Do you know that behaviour?

debsahu, 2018-05-20:
Ha yes. My local version is different only in that aspect, sorry I'll upload the changes version once I get to my computer. What you can do is replace all `if(!taskSendMessage.isEnabled()) taskSendMessage.enableDelayed(TASK_SECOND * 5);` with `taskSendMessage.enableIfNot(); taskSendMessage.forceNextIteration();` through out in server sketch.

toblum, 2018-05-20:
Ah! That explains it. Works great with the change.
I think that I also have problems with the power. The ESP is getting fairly hot and flashing via USB often leads to restarts of the ESP. That could also be a problem. I have to look for a powered USB hub. That makes setup a bit harder for ma at the moment.
I also noticed that the webpage isn't served very reliable, often the ESP just don't returns anything and the browser keeps loading.

LeonVos, 2018-05-25:
Great job DEB !!! I will try a setup this weekend.

thundergreen, 2018-07-04:
Will mesh be implemented now ?

debsahu, 2018-07-04:
Most probably not, but you can try the experimental branch. Other avenues of getting the same results are being explored.

aidbish, 2018-08-04:
@debsahu Hi Debsahu, could you elaborate on what may take it's place? i have a project coming up that would benefit from the mesh style setup.

debsahu, 2018-08-04:
@aidbish ArtNet or some implementation like that.

aidbish, 2018-08-04:
Cool, look forward to it.

aidbish, 2018-08-20:
Hi Debsahu, just a quick question regarding the mesh. i am currently using it till a new version is developed and it does actually work pretty well. what i do a have a problem with is turning them off. for some reason the off doesn't seem to function from the web page.. any ideas?

debsahu, 2018-08-20:
@aidbish which web UI are you using?

aidbish, 2018-08-21:
Im using the index.html one from mclighting experimental, couldn't see one in the meshmclighting folder.

it does look different to the standard one though. i even tried the standard one, but it didnt appear to change

closed 2019-01-24 by debsahu

yashodeepk, 2021-03-09:
@debsahu I am unable to use mDNS with the example provided need a help here