toblum/McLighting #154: Code cleanup

bpohvoodoo, 2018-04-26:
Take a look at my repo at 
https://github.com/FabLab-Luenen/McLighting/commit/c486066c1c507c9d9d8ffea4961c7b12c3305060

maybe you want to to so, too.

debsahu, 2018-04-26:
Yeah excellent idea, during my rewrite I saw the redundancy.

debsahu, 2018-04-28:
@bpohvoodoo made a few changes to mode '~' callback, does send ws and mqtt json reply. Now if users compile with higher bandwidth IwIP2 variant, they will be able to send large 3-4kB MQTT messages. Also noticed my mistake in button mode that you corrected.

bpohvoodoo, 2018-04-29:
Further cleanup
https://github.com/FabLab-Luenen/McLighting/commit/afc9272aae393078f6cf836f8ccefb0d6e1be98a

ChristophA, 2018-05-02:
@bpohvoodoo, just as a feedback, I'm using your latest code with RGBWW stripes without any issues so far.

ChristophA, 2018-07-06:
@bpohvoodoo, will you adopt the latest changes to your RGBW version?

bpohvoodoo, 2018-07-08:
Done ;-)

ChristophA, 2018-07-08:
Cool! THANK YOU, will install it soon!

ChristophA, 2018-07-08:
Mmmh, have you checked in? Latest commit I can see is from 2 month ago...

ChristophA, 2018-07-08:
I had some compiling errors when using the just updated version:
> 'JSON_ARRAY_SIZE' was not declared in this scope
...

Installed ArduinoJSON is version 6.0.1
My settings are:

> #define ENABLE_AMQTT
//#define ENABLE_MQTT
//#define ENABLE_HOMEASSISTANT
//#define MQTT_HOME_ASSISTANT_SUPPORT
//#define ENABLE_STATE_SAVE_SPIFFS
#define ENABLE_STATE_SAVE_EEPROM

The compiling errors are a result of the ArduinoJSON header file: `#include <ArduinoJson.h>` not being including with my settings. With all the home assistant stuff and SPIFFS disabled #include <ArduinoJson.h> is missing.

After adding `#include <ArduinoJson.h>` just after
`#ifdef ENABLE_AMQTT
  #include <AsyncMqttClient.h>`
did fix the compiling errors.

debsahu, 2018-07-10:
@bpohvoodoo I messed up measureJson(). The latest merge should reflect the change.

toblum, 2018-12-16:
Closing issue due to inactivity.

closed 2018-12-16 by toblum