toblum/McLighting #161: New Button on Index.htm

Derodin, 2018-05-04:
Hello,

Have create a New "_mode_" with a "_SetSegment_" Object, but now create a new Bottun on **Index.htm**, but if i click this, triggerd this dont my _"Mode_". can only start this mode with "_http://HOSTNAME/**wohnzimvorn**_"

Here is my Code:
```
server.on("/wohnzimvorn", []() {
      exit_func = true;
      strip.setSegment(0, 120, 180, FX_MODE_STATIC, (const uint32_t[]) {0xff0000, 0x00ff00, 0x0000ff}, 255, false);
      getArgs();
      getStatusJSON();
      #ifdef ENABLE_MQTT
      mqtt_client.publish(mqtt_outtopic, String("OK =wohnzimvorn").c_str());
      #endif
      #ifdef ENABLE_AMQTT
      amqttClient.publish(mqtt_outtopic.c_str(), qospub, false, String("OK =wohnzimvorn").c_str());
      #endif
      #ifdef ENABLE_HOMEASSISTANT
        stateOn = true;
      #endif
      #ifdef ENABLE_STATE_SAVE_SPIFFS
        if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);
      #endif
    });
```

Here is the HTML Code:
```<div class="container mc_pane hide" id="pane2"> (....)
<div class="col s12 m6 l6 btn_grid">
<a class="btn waves-effect waves-light btn_mode_static blue" name="action" data-mode="wohnzimvorn">Wohn Licht: Vorne
<i class="material-icons right">send</i>
</a>
</div>
 (....)
</div>
```

Here for new "**pane Site**"
```
$("#pane0").on("click", ".btn_mode_static", function() {
		var mode = $(this).attr("data-mode");
		var btn = $(this);
		
		wsSendCommand("=" + mode);
		$(".btn_mode, .btn_mode_static").removeClass("red").addClass("blue");
		btn.removeClass("blue").addClass("red");
	});
```

Derodin, 2018-05-04:
Yes, Thanks! its going now :)

debsahu, 2018-05-08:
@Derodin if you want could you fork it and we can link your fork as an example to use segments. I'm closing this as it seems you are able to get this going.

closed 2018-05-08 by debsahu

Derodin, 2018-05-09:
Yes, here is a example:

go to **request_handlers.h**

Search this Line: 
```
#ifdef ENABLE_LEGACY_ANIMATIONS
  void handleSetNamedMode(String str_mode) {
    exit_func = true;
```

and adding this Code under: **exit_func = true;**
```
if (str_mode.startsWith("=wohnzim")) {
      strip.setSegment(0, 145, 155, FX_MODE_STATIC, (const uint32_t[]) {0xffffff, 0xffffff, 0xffffff}, 255, false);
      #ifdef ENABLE_HOMEASSISTANT
        stateOn = true;
      #endif
    }
```