toblum/McLighting #162: Oled Support

gabenthegamer, 2018-05-07:
Is it possible to support oled? Like when its just started it says "Connect to <WifiName>" And when its online it says "Running on <Ip Adress>" And when starting it says <Loading...> If so what oled display do i need to buy?

TechOnlyGamer, 2018-05-07:
that is really a great idea. i saw an esp32 a whole ago with integrated oled. so if thos project gets compatible with it we could see Informationen like ip address or some cool things like that.

toblum, 2018-05-10:
Hi @gabenthegamer,

nice idea. I think the SSD1306 is the best pick. Cheap reliable and supports I2C. I used these before with the Adafruit library https://github.com/adafruit/Adafruit_SSD1306. It shouldn't be too complicated to use these to display status information. I ordered two, will try that out someday later.

Regards
Tobias

gabenthegamer, 2018-05-10:
Thanks! Tell me if you get it to work an what screen ordered

Hilsen Daniel Christensen

toblum, 2018-07-04:
Hi,
I didn‘t find time yet to look into this. The new Displays also did not arrive yet. 
Regards 
Tobias

debsahu, 2018-07-04:
@toblum I have these SSD1306 displays, I can whip something out if this is a priority? Currently busy with Google cloud stuff, once I figure that out I can work on this.