toblum/McLighting #163: Use simple led strip/powerled

rob-v-k, 2018-05-08:
Is it possible to use McLighting with a simple common-anode powerled or a traditional/analog led strip instead of digital WSXX-strips?

debsahu, 2018-05-08:
McLighting is a wrapper around WS2812FX/NeoAnimationFX library. Which in turn are animation libraries that use adafruit's neopixel library and makuna's neopixelbus library respectively. What I would suggest is you look at https://github.com/ItKindaWorks/ESPHelper/blob/master/examples/InputOutput/RGBLight/RGBLight.ino If you want to use with HomeAssistant, I have my own version (not on github yet) that is based on this.

rob-v-k, 2018-05-09:
Thanks for the quick reply. What I find most interesting about McLighting is the possibility to control lights by MQTT, Webinterface and API. It seems the link you posted is only MQTT but at least that's a start.

Keep up the good work.

closed 2018-05-20 by toblum