toblum/McLighting #165: Feature Request: Output for power supply switch

andre1000, 2018-05-10:
It is posible, to integrate a output-pin to switch on/off the power supply of the LED stripe? For example, when the brightness is zero, the power supply will be switched off.

ryancasler, 2018-05-11:
How would you control the power supply?  Your supply has a control input?  Digital or analog? Also, what if your esp is also powered off the same power supply?  Then you can't turn them back on again.

andre1000, 2018-05-11:
The power supply has a digital input to switch it on. The ESP is powerd with a different power supply.

flagstone78, 2018-05-11:
I am thinking about doing this as well. The idle current of WS2812b is quite high when. I measured 10w of which 5 was from ~600 WS2812 and the other 5 from the idling power supply. I am thinking of putting a relay to cut power to the led power supply when they are not lit.

It's pretty easy to code in. Just set a pin to output and set it equal to !(brightness == 0 || mode == OFF) in the main loop.

petertayloruk, 2018-05-11:
If it could be set to send a mqtt message, you could use a sonof power relay. 

⁣Sent from Blue ​

andre1000, 2018-05-11:
The high power consumption in Idle-Mode is why I will switch off the LED string. I will integrate the code in my version.
You will integrate this feature in the next release?
Thank you.

debsahu, 2018-05-11:
As suggested by @flagstone78, Adding `pinMode(_npn_mosfet_pin_, OUTPUT); digitalWrite(_npn_mosfet_pin_, (brightness == 0 || mode == OFF));` before `strip.service();` should work. As to adding this to main release I'm tagging @toblum.

ryancasler, 2018-05-12:
For all those less-than-expert folks out there like me, if you're using an Arduino relay it may have inverse logic.  Just leave off the "!" and it will work perfectly.

aidbish, 2018-07-12:
In the schematic here [link](https://github.com/DotNetDann/ESP8266-MQTT-JSON-SK6812RGBW-HomeAssistant/blob/master/Wiring%20Diagram.png) they use a external MOFSET to turn off the power to the LED strip. They use different type of LED, but wondering if the same idea could be used to reduce power consumption whilst the strip is off

debsahu, 2018-07-12:
Yes that's the idea, a P-Channel MOSFET is ON when the signal is LOW/GND and OFF when signal is HIGH. The choice of MOSFET must be done such that 3.3V generated by io pins of esp8266 are sufficient to trigger the gate of MOSFET, ie look for Vth value on data sheet. In the schematic you posted, they used 5V signal (using a 3.3V to 5V level shifter) to turn the MOSFET gate on or off.

toblum, 2018-12-16:
Closing issue due to inactivity.

closed 2018-12-16 by toblum

ryancasler, 2018-12-20:
> The high power consumption in Idle-Mode is why I will switch off the LED string. I will integrate the code in my version.
> You will integrate this feature in the next release?
> Thank you.

I did some checking....I have two setups that are running off the same power supply.  I put a power measuring outlet on them and with brightness at 0 and 4 wemos d1 minis running, power consumption was 3.1W.  In the US, that works out to <$5 a year.  So, I'm pretty sure that powering the LEDs when they are idle really won't break your bank.