toblum/McLighting #167: Coudnt find "/stripstate.json" Failure!

BassdoxXx, 2018-05-14:
Hi all, I have a problem:
I always get this message, when I try to start:
Should save config
Saving config: {"mqtt_host":"","mqtt_port":"","mqtt_user":"","mqtt_pass":""}WiFiManager config FS Save success!
connected...yeey :)
Open http://192.168.178.118/ to open McLighting.
Use http://McLighting01.local/ when you have Bonjour installed.
New users: Open http://192.168.178.118/upload to upload the webpages first.

**# Coudnt find "/stripstate.json"
Failure!**

Exception (9):
epc1=0xffc48134 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40104412 depc=0x00000000

ctx: sys
sp: 3ffffc20 end: 3fffffb0 offset: 01a0

stack>>>
3ffffdc0: 3fff38fc 3fff3914 3fff38fc 401043c2
3ffffdd0: 0000002e 01820000 c8000000 00000030
3ffffde0: 40103d1d 3ffef350 3ffeea80 3fff2f4d
3ffffdf0: 3fff2f18 3fff2f18 3fff38fc 401044ac
3ffffe00: 4022c373 3fff38fc 3fff3942 4022c37c
3ffffe10: 00000008 3ffffe40 3ffefe08 00000016
3ffffe20: 3ffef76c 00000001 3ffec734 3fff3950
3ffffe30: 3fff38fc 3fff5f48 3fff2f18 40232451
3ffffe40: 005e0001 3ffffb00 3ffffed0 ffffffff
3ffffe50: 00000018 3fff5f48 3fff38fc 40232ed8
3ffffe60: 3ffec000 00000000 53002000 00000000
3ffffe70: 3fff2f1c 00000001 00000000 00000002
3ffffe80: 3fff2e74 00000004 3fff30cc 40233550
3ffffe90: 00000000 00000000 0000001f 3fff5f44
3ffffea0: 00000000 3fff38fc 3fff3968 40232f18
3ffffeb0: 3fff2f18 3ffffee0 00000004 3fff5f44
3ffffec0: 00000016 3fff2f18 3fff5f44 402326a9
3ffffed0: 3fff2f18 3ffffee0 00000004 000000c1
3ffffee0: 00000494 00000000 3fff17dc 3fffdab0
3ffffef0: 3fff5f48 3fff2f18 3fff1a84 00000030
3fffff00: 3fff199c 0000000a 0000000a 00000002
3fffff10: 3fff2e6c 3fff2f18 3fff5f44 40232a48
3fffff20: 402501e6 00000000 3fff38fc 3fff2b30
3fffff30: 4022d3f8 3fff2b34 3ffeac30 4022d406
3fffff40: 40104d7e 00e5b097 00000004 4022d4c4
3fffff50: 3ffeac30 00000000 00000001 4022185d
3fffff60: 40254a28 3fff0468 3fff0490 60000600
3fffff70: 02df79d4 3fff0490 3fff0468 40254a35
3fffff80: 40254a7a 3fffdab0 00000000 3fffdcb0
3fffff90: 3fff04a0 3fffdad0 3fff1a84 402217f7
3fffffa0: 40000f49 40000f49 3fffdab0 40000f49
<<<stack<<<
ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
FS File: /config.json, size: 61B
FS Usage: 76304/957314 bytes

Reading config file... Opened!
Config: {"mqtt_host":"","mqtt_port":"","mqtt_user":"","mqtt_pass":""} Parsed!
WiFiManager config FS Read success!
*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 192.168.178.118
SaveConfig is False!
WiFiManager config FS Save failure!
connected...yeey :)
Open http://192.168.178.118/ to open McLighting.
Use http://McLighting01.local/ when you have Bonjour installed.
New users: Open http://192.168.178.118/upload to upload the webpages first.

**Coudnt find "/stripstate.json"
Failure!**
handleFileRead: /
handleFileRead: /favicon.ico
handleFileUpload Name: /index.htm
handleFileUpload Size: 14287
handleFileRead: /edit.htm
handleFileRead: /

I don't see any pictures and all the links on the webpage won't work (and no colorpickers)
On the Webpage itself I get this error: **"Error on websocket connect."**

I use this settings:
I tried the NodeMCU V3 LOLIN and the NodeMCU Amica
Board: NodeMCU 1.0 (ESP-12E Module)
CPU Frequency: 80MHz
Flash Size: 4M(1M SPIFFS)
IwIP Variant: v2 Higher Bandwidth

Any ideas ?

Thanks in advance.

debsahu, 2018-05-14:
I see 2 issues, firstly you have AMQTT enabled and looks like you are not using it. So my recommendation is comment out these AMQTT and HA lines in `definitions.h`.
```
#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
//#define ENABLE_OTA         // If defined, enable Arduino OTA code.
//#define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
//#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT must be active
#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
//#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
```

Secondly, did you upload https://github.com/toblum/McLighting/blob/master/clients/web/build/index.htm to http://192.168.178.118/upload ?

BassdoxXx, 2018-05-14:
Thank you very much.
I just was too stupid. The ADblocker PiHole is such a big part of my life, that I somethings forget, that it exists :D So I tried it with the guest accesspoint (with nonblocking internet access) and it worked fine ;) 
--->
###"Are connected to internet, if not websocket/color-picker etc wont load?"### 
--->
Was the hint I needed. Thank you once again.