toblum/McLighting #169: Ws2801 LED Stripe possible

Rockiii87, 2018-05-15:
As I'm not familiar with all the different libraries, I kindly want to ask if there is any possibility to use McLighting with my Ws2801 stripes as well. At the moment I'm using them with hyperion and hyperion clone in several places in my house. But I couldn't get it to work with iobroker and Alexa. And as this project is going on very well, I think it's a lot better solution for me.

debsahu, 2018-05-15:
There is a fork of WS2812FX that I also contributed to https://github.com/r41d/WS2801FX even though I do not have WS2801 strips. So you will need to define a clock pin `#define LED_CLOCK_PIN 4` that the clock line of WS2801 is connected and add to definition `WS2801FX strip = WS2801FX(NUMLEDS, LED_PIN, LED_CLOCK_PIN, WS2801_RBG);`. I should also mention that not all animation effects are present. The McLighting version you have to use is [here](https://github.com/debsahu/McLighting/tree/feature/ws2801). I have not kept up with all the code changes in that branch.

Rockiii87, 2018-05-16:
Thanks a lot, i will give it a try this evening.

closed 2018-05-20 by toblum

scheedy, 2019-08-22:
website;  is no longer available.

realhuno, 2019-10-21:
> There is a fork of WS2812FX that I also contributed to https://github.com/r41d/WS2801FX even though I do not have WS2801 strips. So you will need to define a clock pin `#define LED_CLOCK_PIN 4` that [...]

@debsahu can you please re upload the code.... website is no longer available.