toblum/McLighting #170: Having error when compiling and also wantto know about mqtt

Soumojit28, 2018-05-20:
I am having this error

E:\McLighting-master\Arduino\McLighting\McLighting.ino: In lambda function:

McLighting:842: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

McLighting:845: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

exit status 1
'class Ticker' has no member named 'active'

I dont know too much about mqtt, can any one help me how to use mqtt with this and how I can use it to control it over internet and also use it with ifttt to use with google assistance.

debsahu, 2018-05-20:
Update your arduino-esp8266 core to latest release 2.4.1 for Boolean checks for timers.

For IFTTT, use Maker channel and call rest API. For using with Google Assistant, there is no direct way, what I do is have it connected to Home Assistant and that has a Google Assistant & Alexa & HomeKit components that can be used in parallel.

debsahu, 2018-05-20:
Check the wiki to updating software:  https://github.com/toblum/McLighting/wiki/Software-installation

MQTT is just like text messaging. mqtt_hostname is like a phone number, so every mqtt_hostname MUST be unique. The default hostname is set in line 10: const char HOSTNAME[] = "McLighting01";.

Secondly topics, topics are like group chats. If 2 clients want to read and respond to the same message, they need to "subscribe"  to the SAME topic. In McLighting, users can't choose the topic and is set automatically using the HOSTNAME set in line 10. 

In MQTT, one can also "publish" on a topic equivalent to sending a text to the group. Once a new message is received on server, all clients "subscribed" to the topic will receive the message.

Again for all this you will need a MQTT server! If you have a Raspberry Pi ($35), you can install mosquitto MQTT server to act as a MQTT server.

Soumojit28, 2018-05-21:
I successfully compiled and upload the code in nodemcu, the web interface is working, but the leds not changing any color/ no changes in led, I used pin D5 in code, can you please help?  (Its working fine with previous version of the mc lighting)

debsahu, 2018-05-21:
Could you recompile with `#define PIN 14` in definitions.h. And are you using WS2812B?

Soumojit28, 2018-05-21:
Tried but no luck with it

debsahu, 2018-05-21:
What are the options you enabled? Did you enable NeoAnimationFX or are you using default WS2812FX? And Serial output would be useful.

Soumojit28, 2018-05-21:
http://prntscr.com/jkrfzk

I am using NeoAnimationFX cause WS2812FX is giving some kind of error with the new version code.

getting some weird data in serial monitor  http://prntscr.com/jkri7p

But its connecting to my wifi and the web interface is working fine

debsahu, 2018-05-21:
Ahh, NeoAnimationFX will ignore pin number. By default it will use PIN D9/RX/GPIO3 unless you change the method to UART which will use PIN D4/GPIO2. Those are the only two choices you have due to hardware limitation of ESP8266.

Soumojit28, 2018-05-21:
Thanks its working, I have made some cool things with this (Previous Version)- checkout those images
 
Led Hoddie with wifi control - https://imgur.com/taKhcUo ,  https://imgur.com/i2IBTPK

Led fedora hat with li-ion battery and wifi control from phone - https://imgur.com/QQ4Dbyf
I have tutorial for making those and I used mc lighting in it (Maybe U guys can showcase it) - https://www.instructables.com/id/WiFi-Led-Fedora-Hat-ESP8266-WS2812b/

Now, I want to add some new features in my project, like ifttt support, As I said before I dont know too much about mqtt, I have a adafruit.io account, can I use it to control? If yes than how?

toblum, 2018-05-21:
@Soumojit28 BTW: Nice projects!
Linked your video in the McLighting project video collection: https://www.youtube.com/watch?v=tVZ9yOzEAk8&list=PL3RrveYYJCuRDhrZXcRSQgtsIZ2lH64bv

Best regards
Tobias