toblum/McLighting #173: STATIC LEDs are Crazy.

carlymx, 2018-05-31:
Hello

I have acquired a nodeMcu module ESP8266 esp-12e and a strip of leds WS2812B / M of 4mts 60leds / m.

I followed the detailed steps in the WIKI and the videos religiously and at first everything works well less for a detail.

The LEDs become "crazy". You order a static color and it takes a while to react and then they begin to blink, you put another color and they do the same ... putting an effect is completely impossible since it does not look anything like what would have to happen.

I have tried a thousand times to load again, reinstall all libraries ... with 10/15/20/80 leds and always do the same, changed the power supply.

I've also tried to change the Digital IO, and I've tried the sketch blick which works perfectly ...

Please help me. What I can be doing wrong...?

[image]

[image]

[image]

debsahu, 2018-06-02:
You need a good power supply, I suspect there is lots of noise in your 5V power line. Usually adding a few different capacitors in parallel would resolve this issue. Also are you using a level shifter to send data?

petertayloruk, 2018-06-02:
> [![" VIEW GIF HERE

What amp power supply do you have? 

⁣Sent from Blue ​

carlymx, 2018-06-08:
Thank you very much for answering.

When I read the comments I realized that I was using a 9V 200mA power supply with a [voltage regulator at 5V and 3.3V](https://es.aliexpress.com/store/product/MB102-Breadboard-Power-Supply-Module-3-3V-5V-For-Arduino-Solderless-Breadboard/1414081_2038557602.html?spm=a219c.search0104.3.1.5cb01810ks4tuf&ws_ab_test=searchweb0_0,searchweb201602_4_10152_10151_10065_10344_10068_10342_10547_10343_10340_10548_10341_10696_10084_5722520_10083_10618_10307_10820_10301_10821_10303_10059_5722620_5722920_5722720_5722820_100031_10103_10624_10623_10622_10621_10620,searchweb201603_25,ppcSwitch_5&algo_expid=96307a3c-ec58-41f2-ae6a-789b8652566d-0&algo_pvid=96307a3c-ec58-41f2-ae6a-789b8652566d&priceBeautifyAB=0).

Provide a PC source with direct supply of the yellow cable (5V) to the LED strip and the 3.3v controller for the ESP8266 and the same thing happened to me ... It could not be, a PC source giving problems ...

I tried with several chargers for mobile 5V-2.5A and 5V-1A with total success, check all the ESP8266 models that I have and without problems.

[image]

Saying that 22 leds + the ESP module consume an average of 220mA as shown by the Ammeter.

[image]

Thank you very much for the help.

closed 2018-06-08 by debsahu