toblum/McLighting #174: Loss of wifi connection after some hours

ChristophA, 2018-06-01:
Hi,
I'm using the RGBW version #154 @bpohvoodoo build based on this wonderful project with 4 Wemos D1 mini Pro. Thanks to all for you contribution!

The only issues I have so far is the loss of the wifi connection after some hours. The connections also drops when using the reset button on the D1 mini Pro or pulling pin RST to ground. It always ended up in AP mode and I had to reconnect to the wifi manually.

I haven't figured out what the root cause is, but it looks like I've found a workaround for an automated reconnection. In the main sketch (McLighting.ino) in the wifiManager section (after row 303) I've added:
`//hard coded wifi settings`
`wifiManager.autoConnect(APNAME, APPWD);`

I've added the variables in definitions.h:
`//hard coded wifi settings`
`const char APNAME[] = "ssid";`
`const char APPWD[] = "wifi_password";`

Now a reset ends up in a reconnected D1 mini Pro ;-)

> WiFiManager config FS Read success!
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.188.38
*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.188.38
SaveConfig is False!
WiFiManager config FS Save failure!
connected...yeey :)

Anyone having similar issues? Anyone an idea if there's a better solution? Looks like now there are 2 tries to AutoConnect. Maybe the build-in standard one is not working properly!?

Again, great project, thank you!

Cheers,
Christoph

debsahu, 2018-06-01:
What branch of wifimanager are you using?

ChristophA, 2018-06-01:
I'm using WifiManager by tzar Version 0.12.0. Should be the master branch.

debsahu, 2018-06-01:
Ok that is the branch I recommend, could you erase your flash before uploading (Arduino -> Tools -> Erase Flash -> All Flash Content). From numerous issues open on wifimanager's GitHub, this issue could happen when flash memory gets corrupted which is very common when using baud rate over 115200 for flashing, using cheap USB cable or using esp8266-arduino 2.4.xx core (no workaround the last one).

debsahu, 2018-06-01:
To set static IP I would suggest you look into https://github.com/tzapu/WiFiManager/blob/master/README.md#custom-access-point-ip-configuration It's IP, Gateway and Subnet.

The master wifimanager version has no input for static DNS (used for NTP etc), but there is no use that here in McLighting. I wrote the static DNS input for wifimanager which can be found in develop version.

debsahu, 2018-06-02:
To hard code a WiFi setting, add `WiFi.begin("wifi_SSID","wifi_password");` before wifimanager.autoconnect and `bool shouldSaveConfig = false;` shouldn't be modified.

relax81, 2018-06-03:
Thank you, i was just looking for something similar. I've used this library to build a status light for my parents, so they've also a visual effect in the living room when the door bell goes on. 
The only problem was that the light would go into the wifimanager AP setup mode if any wifi issues happened. Re-configuring is too much for them (they're 70+).

I think hardcoded Wifi Settings would be a nice thing in general, at least as an option.

ChristophA, 2018-06-04:
Even with fixed IP address and hardcoded wifi settings the wifi connection drops over night. It always ends up in `connection lost` in the debug log.Short push on the reset button of the D1 mini Pro properly re-connects.

I'm now testing the original McLighting version with just a basic setup to see if it more stable and If yes there's an issue with the RGBW version.

debsahu, 2018-06-04:
Tagging @bpohvoodoo

relax81, 2018-06-04:
I've now experienced as well that the "hardcoded" version doesn't solve my issue. Today some wifi issues happened and the whole system switched back to AP mode.  I wish I could avoid that and have a way that it just tries re-connecting to the old wifi settings until they're available again

bpohvoodoo, 2018-06-06:
I haven't seen this problem at my devices so far. I'll have to inestigate.

ChristophA, 2018-06-17:
Sorry for the late update. I've been on a business trip, so further testing has been delayed.

I've tested both, the original and the RGBW version on 4 Wemos D1 mini Pro. I had always the same issue: 'connection lost'.

Meanwhile there was an update for my wifi router and I've completely setup a fresh Arduino 1.8.5 environment from scratch. The RGBW version is now running for 2 days without dropping connection. So it looks like the wifi connection was the issue.

Thank you very much for all the support and testing!

closed 2018-06-17 by ChristophA