toblum/McLighting #175: SKETCH_VERSION "2.1.1" does not fit in the ESP-01 memory

carlymx, 2018-06-02:
Hi.

I am trying to compile Sketch v2.1.1 on an ESP8266 ESP-01 and it tells me that it does not fit. By selecting a minimum of 512K (64K SPIFFS) in order to upload "index.html" the memory usage is 104%.

[image]

Disabling some things that I do not need in "definitions.h" I get down to 101%.

```
//#define USE_NEOANIMATIONFX  // Uses NeoAnimationFX, PIN is ignored & set to RX/GPIO3 or UART method: D4/GPIO2, see: https://github.com/debsahu/NeoAnimationFX
#define USE_WS2812FX          // Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX

// Neopixel
#define PIN 0           // PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 22       // Number of leds in the strip
//#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
//#define BUTTON 4         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

const char HOSTNAME[] = "McLighting01";   // Friedly hostname

#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
//#define ENABLE_OTA         // If defined, enable Arduino OTA code.
#define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
//#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT must be active
//#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
//#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
#define ENABLE_LEGACY_ANIMATIONS
```

Do you think you can purge more?

I think it would be interesting to be able to use this project again in ESP-01.

Thank you.

debsahu, 2018-06-02:
Disable OTA, I don't think you will have enough space to OTA anyways.

debsahu, 2018-06-02:
Also have you seen this: https://youtu.be/xyc1gCjguRU very easy to do.

closed 2018-06-06 by toblum