toblum/McLighting #176: CHange green with blue

thundergreen, 2018-06-05:
hi.. i got everything up and running. using USE_WS2812FX   for my strip but green and blue are inversed. How do i fix this? sorry for dumb question :)

debsahu, 2018-06-05:
Change `WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_GRB + NEO_KHZ800);` to `WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_BRG + NEO_KHZ800);`

thundergreen, 2018-06-06:
in wich file? cannt find it in definitions.h

closed 2018-06-06 by toblum

Gintazas, 2019-03-24:
hi every one. i am new on arduino espacialy on led control. i use w2812b led strip. upload done, but in Mc Lightning UI (when strip start) is green, but have to be red. Can't find where replace color green to red. 
Other moments: strobe rainbow(27) mode on Mc Lightning UI have only strobe effent only - isnt rainbow together
TV(TV) and E131(E131)no reaction.
How to add to CUSTOM effect?
Thanks for help