toblum/McLighting #177: Occasionally resets of brightness.

davidblumer, 2018-06-05:
I'm using the REST-Api to turn my LEDs on and off. To toggle on i'm calling "/rainbow" and for off "/off".
At some random times the strip is getting powered and displays rainbow mode.

davidblumer, 2018-06-06:
Sometimes at random times the strip is turning on again without me calling the api.

toblum, 2018-06-06:
HI @davidblumer ,

these ransom reboot often happen because of not 100% stable power supplies. The ESP is quite sensitive to power supply. What is your setup (how many LEDs, power supply, board, ...)? 
Have also a look here: https://learn.adafruit.com/adafruit-neopixel-uberguide/powering-neopixels

Regards
Tobias

davidblumer, 2018-06-26:
I bought a new power supply 5V 8A - same one i use for my ambilight without any problems. Reboots still happen randomly. I got 3 meters 2812b / 60 - so ~ 180 LEDs. My board is a ESP8266 NodeMCU.

synssins, 2018-07-17:
I also have random reboots on every single unit I use with McLighting, or random drops back to solid blue lights until I reset the device. It is still connected to wifi when the latter happens.

My power supplies are rock solid, tested on an oscilloscope, metered for output to verify 5v, etc. In one case, I'm running 48 LEDs, and still encountering the issue on a 4 amp power supply.