toblum/McLighting #180: TODO: Update JSON code to ArduinoJSON 6.xx

debsahu, 2018-06-12:
For some reason, Arduino IDE is updating ArduinoJSON to 6.xx version. This means McLighting won't be compatible, manually downgrade to [ArduinoJSON to 5.13.2](https://github.com/bblanchon/ArduinoJson/releases/download/v5.13.2/ArduinoJson-v5.13.2.zip) until we fix the code.

toblum, 2018-06-13:
Hi @debsahu,

thank you for investigating. I have no McLIghting installation available this week to try out but I will think I may find time next week.

Best reagrds
Tobias

debsahu, 2018-06-17:
This [push](https://github.com/toblum/McLighting/commit/e499033ef33c078d951b80893885fe781cc81270) should address this issue. Needs testing before merging with master.

toblum, 2018-06-19:
Hi @debsahu,

thank you for fixing this. More changes than I expected.

I've put this version on my testing setup. It seems to work flawlessly. The only thing I've noticed was a restart/hangup after some hours, but I had this with the experimental branch too, so it seems not to be related with these changes, I guess. I didn't experience these restarts before with this setup (Core 2.4.1, NodeMCU board, 24 pixel ring, no caps, resistors). Message in the console was:
" ets Jan  8 2013,rst cause:4, boot mode:(1,7)
wdt reset"

When you also think that this is not related, we could merge the push.

Best regards
Tobias

debsahu, 2018-06-19:
ArduinoJSON changes are extensive and it's a good thing as the library is extremely light, written in same template manner as NeoPixelBus.

Regarding resets: I'm thinking ws2812fx seems to be the issue, can't pin point exactly. Someone on gitter of also experiencing the resets as well, by just running rainbow mode for 6 hrs or so. My test rig is stuck in another project for a while now and the other one is using NeoAnimationFX and not showing any issues.

toblum, 2018-06-19:
Hi @debsahu,
I merged your changes in dev branch, along with some minor changes from me, to the master. I hope everything is fine.

I couldn't reproduce the restarts today, I will have a look at it.

Regards
Tobias

closed 2018-06-19 by toblum