toblum/McLighting #185: Help wanted

pradhyuman18, 2018-06-18:
How to integrate it with Google assistant via webhook?(so we can work it as like as Siri as shown)

debsahu, 2018-06-18:
Not straight forward. The most direct way is to use dialogflow. The way I use it is integrate it with home assistant and have everything on home assistant integrated to Alexa, Siri and Google.

pradhyuman18, 2018-06-20:
Another way ? Please

debsahu, 2018-06-20:
Another approach would be configure IFTTT ( maker_webhook ) to invoke webhook. This will require you to expose your esp8266 to the internet (not safe). IFTTT can be easily integrated with Google Assistant. This is the last resort I would recommend, other ideas welcome here.

ryancasler, 2018-06-20:
Yes, IFTTT can use Google Assistant as an trigger but not take actions on
you're Google Assistant. You cannot, for example, use IFTTT to trigger a
broadcast on your Google home.

cjs1502, 2018-06-20:
can anyone explain me how to do it?

debsahu, 2018-06-20:
Let me point to the easiest approach: McLighting > Home Assistant > Google Assistant.

1. Setup Home Assistant, resources: https://www.home-assistant.io/getting-started/ which can be on the same RPi as the MQTT server
2. In McLighting, enable Home Assistant auto-discovery by uncommenting `#define MQTT_HOME_ASSISTANT_SUPPORT` to automatically add McLighting to Home Assistant.
3. Setup Home Assistant's Google Assistant as per instructions here: https://www.home-assistant.io/components/google_assistant/
4. Profit!
5. Easter egg: Siri can be easily integrated as well https://www.home-assistant.io/components/homekit/ and Alexa as per instructions here: https://www.home-assistant.io/components/emulated_hue/

cjs1502, 2018-06-20:
is it possible without home assistant?
as i dont have any access to a raspberry pie right now

debsahu, 2018-06-20:
It is possible but not as easy and straightforward.

1. Expose your ESP8266(McLighting) to internet
    a. Expose port 80 from outside (internet) to local port 80 of ESP8266's IP
    b. Or use ngrok (more details here: https://ngrok.com/ )
2. Use IFTTT webhooks: https://ifttt.com/maker_webhooks
3. Create an app where a GET request to the internet exposed address using McLighting's RESTful API (https://github.com/toblum/McLighting/wiki/HTTP-REST-API)
4. In IFTTT, setup google assistant and integrate with webhooks

cjs1502, 2018-06-20:
how do expose my esp8266 without download anything?

debsahu, 2018-06-20:
Third way is to use diagflow https://dialogflow.com/ which also requires one to expose the ESP8266 to the internet. Instructions are here: https://dialogflow.com/docs/getting-started , which can be adapted to call a McLighting RESTful API.

debsahu, 2018-06-20:
@cjs1502 Go to your router settings and expose the desired port.

cjs1502, 2018-06-20:
this is so confusing,i am  not doing it.but thank you for your help.

debsahu, 2018-06-20:
@cjs1502 One can thank Google for that!

pradhyuman18, 2018-06-21:
How can I use ifttt webhooks with it?

ryancasler, 2018-06-21:
You can't.  Webhooks can be triggered by Google home via the IFTTT integration but you can't trigger Google Home actions from Webhooks.

pradhyuman18, 2018-06-21:
I know but what to put in body section in webhooks?

ryancasler, 2018-06-21:
I don't think you understand.  Assistant relay has nothing to do with Webhooks.  You cannot use Webhooks to trigger Google Assistant. Assistant can trigger a we hook to something else.  So, what is the something else you are trying to trigger.  And don't say relay cause I just told you it doesn't work that way.

cjs1502, 2018-06-21:
it would be great if any of the contributor would make a video showing how to setup google assistant with mc lighting.

ryancasler, 2018-06-21:
It would all depend on what "man-in-the-middle" system you are using.  I can control my McLigting setups through Google Assistant.  I have them set up through SmartThings.  But to do a write up on that setup would take me hours because it took me days to figure out.  Plus, if you don't like McLigting, go find another solution.  Stop pestering for more more more more.  It's free damn it.  Take it and say thank you. LOL

cjs1502, 2018-06-22:
this not about getting it done without spending any money,this is about the joy that you get it from making it. you wont understand that little joy we people get even after using that simple example of blink that would light up that onboard led.

ryancasler, 2018-06-22:
I don't?  You know nothing about me.  Please don't presume.  And you're not making something.  You're asking for a custom solution for free.

cjs1502, 2018-06-22:
i am not presuming anything. i am just saying that you being a contributor should be helping us like @debsahu is doing rather than giving rude comments.

ryancasler, 2018-06-22:
You're not presuming you're just telling me what I should be doing? I see.

cjs1502, 2018-06-22:
i am also not telling you what you should be doing,but you should be helping beginners as you are a pro at this kind of stuff, and helping is totally your choice , no one is forcing you to that.this conversation will never end this will go on forever. i will stop here. and thank you for all your contributions in this git.

ryancasler, 2018-06-22:
Bit you're not asking for help. You're asking new features to be implemented. New features, which, by the way, are totally outside the scope of this project.  You'd have to write a program using the Google Assistant SDK. Something that doesn't even run on an ESP board.  You don't even realize how entitled you're being so you?  We should just do all that work for you because you asked?  Are you kidding me right now?

debsahu, 2018-06-22:
Let's all calm down. There is nothing wrong in asking for a tutorial as I understand there are a few newbies here. Let's be inclusive and try to be respectful here.

I do make YouTube videos, but on things that I build or figure out something new, be it hardware or software. I already have a few things planned for the next few uploads, and can make a tutorial on integrating McLighting with various assistants later (no promises). Even if I make the tutorial, it will be one of the many ways to do it and may not suit your needs. Like @ryancasler mentioned there are many many man-in-the-middle solutions, dialogflow (formally API.ai), IFTTT, Home Assistant, Artik Cloud, Adfruit.io, AWS IoT, aREST, etc just to make a few.

cjs1502, 2018-06-22:
i never asked for a new feature to implement, i just asked for a tutorial to use google assistant on McLightings , as home assistant is already there in it.
@debsahu i saw your youtube channel the other day, you are doing good, i hope you upload new contents that include McLightings
looking forward to it.
and thank you for being a good sport, i appreciate that.

sujitrp, 2018-09-22:
@ debsahu :  I am looking for following architecture to voice control my setup.
My set up is working good with Google home and Alexa using raspberry Pi -Openhab configuration.
to remove the Pi & openhab for voice control following architecture will work safe . 
being new bee on coding side last few days I am finding it difficult to  enable Adafruit MQTT.
appreciate your guidance on the same

[image](https://user-images.githubusercontent.com/7476632/45917754-e40b0000-be99-11e8-950b-e151ab5824df.png)

sujitrp, 2018-09-24:
Now I am sure I am lost in McLightining code while enabaling google assistance using adafruit - Ifftt - Google assistance approach : 

Need help to remove ' failed to connect RC=5 error ' .  

What i did till now to make the existing program compatible with adafruit MQTT : 
1. Update the host name with <Userid>/feeds  (quick check for mqtt . (Original plan is to modify Mqtt_ha topic. )
2. hard coded the Adafruit MQTT broker  
              broker : io.adafruit.com
              Port : 1883
              username : openhab
              AIO Key: <key>
3. Soft entered the Adafruit mqtt credential (Post Booting uisng Web server) 

(On My local mosquitto Broker it works & get connected )
all the cases on my serial monitor post successful Wi-fi connection ans IP assignment  I got " MQTT failed to connect RC =5"

Now next few debug steps I am not sure I did right or wrong 
Post google I understood that Mqtt Port issue while connecting adafruit mqtt 

so I changes the Mqtt_port propertied to uint8_t mqtt_port[6] = "" from char
Post of which I got multiple times error in verify .
so I replaced atoi() some place and update atoi() at some places .  I am getting 'exit status 1
invalid conversion from 'uint8_t* {aka unsigned char*}' to 'const char*' [-fpermissive]' errro 
full verify log : 
Arduino: 1.8.5 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, 4M (1M SPIFFS), v2 Higher Bandwidth, Disabled, None, All Flash Contents, 115200"

C:\01SujitDdrive\OS\arduino-1.8.5-set3\arduino-builder -dump-prefs -logger=machine -hardware C:\01SujitDdrive\OS\arduino-1.8.5-set3\hardware -hardware C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages -tools C:\01SujitDdrive\OS\arduino-1.8.5-set3\tools-builder -tools C:\01SujitDdrive\OS\arduino-1.8.5-set3\hardware\tools\avr -tools C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages -built-in-libraries C:\01SujitDdrive\OS\arduino-1.8.5-set3\libraries -libraries C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\sketchbook\libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss1460,Debug=Disabled,DebugLevel=None____,FlashErase=all,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\spp07561\AppData\Local\Temp\arduino_build_846746 -warnings=none -build-cache C:\Users\spp07561\AppData\Local\Temp\arduino_cache_679805 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\tools\esptool\0.4.13 -verbose C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\sketchbook\McLightingadafruit\Arduino\McLightingadafruit\McLightingadafruit.ino
C:\01SujitDdrive\OS\arduino-1.8.5-set3\arduino-builder -compile -logger=machine -hardware C:\01SujitDdrive\OS\arduino-1.8.5-set3\hardware -hardware C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages -tools C:\01SujitDdrive\OS\arduino-1.8.5-set3\tools-builder -tools C:\01SujitDdrive\OS\arduino-1.8.5-set3\hardware\tools\avr -tools C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages -built-in-libraries C:\01SujitDdrive\OS\arduino-1.8.5-set3\libraries -libraries C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\sketchbook\libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss1460,Debug=Disabled,DebugLevel=None____,FlashErase=all,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\spp07561\AppData\Local\Temp\arduino_build_846746 -warnings=none -build-cache C:\Users\spp07561\AppData\Local\Temp\arduino_cache_679805 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\tools\esptool\0.4.13 -verbose C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\sketchbook\McLightingadafruit\Arduino\McLightingadafruit\McLightingadafruit.ino
Using board 'd1_mini' from platform in folder: C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\hardware\esp8266\2.4.2
Using core 'esp8266' from platform in folder: C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\hardware\esp8266\2.4.2
Detecting libraries used...
"C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\spp07561\AppData\Local\Temp\arduino_build_846746/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=1460   -DA [... cut ...]
Using library Adafruit_NeoPixel at version 1.1.6 in folder: C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\sketchbook\libraries\Adafruit_NeoPixel 
Using library Ticker at version 1.0 in folder: C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\hardware\esp8266\2.4.2\libraries\Ticker 
Using library Hash at version 1.0 in folder: C:\01SujitDdrive\OS\arduino-1.8.5-set3\portable\packages\esp8266\hardware\esp8266\2.4.2\libraries\Hash 
exit status 1
invalid conversion from 'uint8_t* {aka unsigned char*}' to 'const char*' [-fpermissive]

Now I am sure I am lost .
Need help to get read of failed to connect RC=5 error .

sujitrp, 2018-09-24:
Hi Finally i am connected to Io.adafruit.com by  changing
line 352 to 'strcpy(mqtt_port, custom_mqtt_port.getValue());'
now next issue  MQTT connection gets disconnected frequently .. Please find the log 

strcpy(mqtt_port, custom_mqtt_port.getValue());

debsahu, 2018-09-24:
@sujitrp Using Adafruit MQTT with pubsubclient/amqtt methods are not straight-forward, you need to use adafruit mqtt code. You need to first figure that thing out before trying McL.