toblum/McLighting #186: Missing stripstate.json

SirJMD, 2018-06-18:
I followed the "Software installation" and got it connected to my wifi. After that I get this error via the serial monitor:

```
Open http://192.168.1.107/ to open McLighting.
Use http://McLighting01.local/ when you have Bonjour installed.
New users: Open http://192.168.1.107/upload to upload the webpages first.

Couldn't find "/stripstate.json"
```

And sadly I don't get any UI after uploading the index.htm. So I'm stuck at the missing stripstate.json file.

debsahu, 2018-06-18:
Initially there is no stripstate.json file, so seeing that is normal. Once you turn a particular mode on, it will create this file and read from it in consecutive reboots. Did you upload https://github.com/toblum/McLighting/blob/master/clients/web/build/index.htm ?

SirJMD, 2018-06-18:
I must be beyond tired (alone with sick kid), because I uploaded the \Arduino\McLighting\data\index.htm
So yea.. after I uploaded the correct one, it of course works. Thanks!

If I want to change the number of LEDs, how do I do that without re-uploading the code?

debsahu, 2018-06-18:
no way to change the number of LEDs on the fly with current code, sorry have to reupload.

closed 2018-06-18 by debsahu