toblum/McLighting #187: can you configure static ip?

arturodiazad1, 2018-06-18:
I make it work with homebridge and it frequently changes ip, do you know if there is any option to configure the static ip?

debsahu, 2018-06-19:
Add `wifiManager.setAPStaticIPConfig(IPAddress(192,168,1,128), IPAddress(192,168,1,1), IPAddress(255,255,255,0));` before `if (!wifiManager.autoConnect(HOSTNAME)) {`. The order is IP, gateway and Subnet.

toblum, 2018-06-19:
@debsahu Maybe we should add that line commented out in the code. It's asked quite often. What do you think.

debsahu, 2018-06-19:
@toblum yes I agree, also add a wifimanager timeout line and when we decide to switch to be webUI, we should remove the esp.restart() from wifimanager.autoconnect() to provide offline access.

toblum, 2018-06-19:
I added the example line to the code. I noticed that it has to be
"setSTAStaticIPConfig" instead of "setAPStaticIPConfig" in this case.

closed 2018-06-19 by toblum

debsahu, 2018-06-19:
As @toblum mentioned setSTAStaticIPConfig is the correct way to set custom static station IP, not setAPStaticIPConfig.