toblum/McLighting #188: How to added new

ali1977, 2018-06-19:
How I can added a new function in MClighting.
I have this function and I don't found the right way.

```
void runway() { // Landebahn_test
  static int j=0, q = 0;
  static boolean on = true;

for (uint16_t j = 0; j<=11; j=j+3)                       
{
  strip.setPixelColor(j, strip.Color( 255,255, 255));
  
  }
for (uint16_t j = 12; j<=15; j=j+1)                       
{
  strip.setPixelColor(j, strip.Color( 255,0, 0));
  
  }
for (uint16_t j = 16; j<=24; j=j+3)                       
{
  strip.setPixelColor(j, strip.Color( 255,255, 255));
  
  }
for (uint16_t j = 25; j<=29; j=j+1)                       
{
  strip.setPixelColor(j, strip.Color( 0,255, 0));
  
  }     
      strip.show(); // display
   
  lastUpdate = millis(); // time for next change to the display   
 
}
```

debsahu, 2018-06-19:
For more details see WS2812FX custom effect example [here](https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/ws2812fx_custom_effect/ws2812fx_custom_effect.ino).

Add this to setup()
```
strip.setCustomMode(runway);
```
and add this anywhere in the code
```
uint16_t runway() { // Landebahn_test
  static int j=0;
  for (uint16_t j = 0; j<=11; j=j+3)
    strip.setPixelColor(j, strip.Color( 255,255, 255));
  for (uint16_t j = 12; j<=15; j=j+1)
    strip.setPixelColor(j, strip.Color( 255,0, 0));
  for (uint16_t j = 16; j<=24; j=j+3)
    strip.setPixelColor(j, strip.Color( 255,255, 255));
  for (uint16_t j = 25; j<=29; j=j+1)
    strip.setPixelColor(j, strip.Color( 0,255, 0));
  return 20; // return the delay, in this example returns 20ms
}
```

When mode FX_MODE_CUSTOM or 56 is called your function runway is called.

ali1977, 2018-06-20:
sorry, but i became this error
'runway' was not declared in this scope

and how can add this allso in the app?

debsahu, 2018-06-20:
Copy the contents from `runway()` I posted into the end of `colormodes.h`. Also more specifically add the `strip.setCustomMode(runway);` in between `strip.init();` and `strip.begin();`

closed 2018-07-10 by debsahu

vankipish, 2019-08-05:
sorry, but it all realy doesn't work. I've got message like this:
colormodes.h:104:11: error: request for member 'setPixelColor' in 'strip', which is of pointer type 'WS2812FX*' (maybe you meant to use '->' ?)....
how can i add my costom effect? is it way actual, may be, and problem in my curve hands? but i've tried by instructions