toblum/McLighting #189: Exception (9)

SirJMD, 2018-06-19:
Yesterday, with a little help, I got McLighting running. I then wanted to change the number of LEDs and nothing else, and for some reason I now get the following error:

```
Exception (9):
epc1=0x4020f924 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000007 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  40251d80 3ffef8b4 3fff0830 00839762  
3fffff80:  40251da6 3fffdab0 00000000 3fff09b0  
3fffff90:  3fffdc10 00000000 3fff38f4 40255d28  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

My background is electronics, not programming, so I'm not sure if this is a software or hardware issue based on the error report above. I've tried a few example sketches, without encountering any issues.

I've followed the "Software installation", installing the known working version of the needed libraries.

toblum, 2018-06-19:
HI @SirJMD,

from the exception message it's hard to tell what happens. There is a ExceptionDecoder tool that helps finding the reason: https://github.com/me-no-dev/EspExceptionDecoder 
When you have installed it (just copy it into the described place and restart Arduino ESP) you can open it from the Tools menu. Then paste the stack into the window. Then send back the decoded message. That would help alot.

Regards
Tobias

SirJMD, 2018-06-19:
Version 1.1.0 seems to have an issue that isn't fixed yet. The previous version seems to run. The stack I'm getting:

```
>>>stack>>>
3fffff70:  40251d80 3ffef8b4 3fff0830 00a154fb  
3fffff80:  40251da6 3fffdab0 00000000 3fff09b0  
3fffff90:  3fffdc10 00000000 3fff38f4 40255d28  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<
```

And the decoded result:
```
Decoding 5 results
0x40251d80: ets_timer_handler_isr at ?? line ?
0x40251da6: ets_timer_handler_isr at ?? line ?
0x40255d28: wifi_set_status_led_output_level at ?? line ?
```

It seems incomplete.

toblum, 2018-06-19:
Hi @SirJMD,

which version of the ESP core are you using?

Regards
Tobias

SirJMD, 2018-06-21:
I'm running "ESP8266 Community" v. 2.4.1, same as in the guide.

ryancasler, 2018-06-22:
What board are you using?

SirJMD, 2018-06-22:
I'm using a D1 mini (ESP8266MOD). Brand new out of the anti-static bag. Same error.

I downloaded the new version of McLighting that was uploaded a few days ago. Getting a lot of json errors, so that one is no good. Also the definition of pin D1 results in errors for me. I have to use pin number (i.e. 2), in order to not get errors.

With a brand new McLighting project, with only "pin D1" changed to "pin 2", I can successfully compile, upload a get the webserver to start. No crashes.

MQTT enabled and AMQTT disabled is also fine.

SirJMD, 2018-06-22:
I think I found the error. I had enabled these two, thinking I needed both (not sure what the difference is..):
#define MQTT_HOME_ASSISTANT_SUPPORT
#define ENABLE_HOMEASSISTANT

And the check further down in the code has an error, so it allows both to be enabled.

```
#if ( !defined(ENABLE_HOMEASSISTANT) and defined(MQTT_HOME_ASSISTANT_SUPPORT) )
#error "To use HA support, you have to either enable Homeassistant component"
#endif
```

Again, I'm an electrical engineer, not software engineer, but.. isn't there a exclamation sign missing there?

ryancasler, 2018-06-22:
Remember that Pin 2 is actually D4, which is connected to the Bulletin LED
near the antenna.  I haven't had good luck using that pin for McLighting as
I think the onboard LED is causing some type of interference or current
drain in the signal.  I would use D1 (GPIO 5) D2 (GPIO 4) or 5-7 (GPIOs 14,
12, 13).  All of the other pins have special functions that could screw
with McLighting.  You only have to use one, so why not avoid those, just in
case (IMHO).

toblum, 2018-06-24:
@SirJMD I'm not sure with the exclamation mark. I think this line is only to check if ENABLE_HOMEASSISTANT is also active when you're using MQTT_HOME_ASSISTANT_SUPPORT. @debsahu is this correct?

Regarding the pins I would now suggest GPIO 2 or 3 because that way you could easily swap to the great NeoAnimationFX library.

debsahu, 2018-06-24:
`#define ENABLE_HOMEASSISTANT` is required if you plan to use with Home Assistant (HA). `#define MQTT_HOME_ASSISTANT_SUPPORT` requires HA competent to be enabled, what this is a automated way to add McLighting to home assistant. This requires home assistant discovery competent to be enabled and the details are in McLighting wiki. The check I have in there is correct, to use the automated discovery of McLighting in HA, HA component needs to be enabled.

SirJMD, 2018-06-25:
@ryancasler That is true. I haven't encountered issues with D4, but changed it to D2 just to be sure :)

@toblum I'm using D2 (GPIO4) now and it seems to work fine. Some times when I upload the sketch, I get the same error as posted in the first post. So far I haven't been able to find the cause of this. I've tried no less than 5 different D1 mini boards, to rule out hardware issues.

Can the flash settings be to blame? I've tried many different settings to try and force the fault, but haven't found a clear indication, other than 4M with 3M SPIFFS is no-go.

@debsahu Isn't enabling MQTT_HOME_ASSISTANT_SUPPORT putting me in a catch 22? If I enable ENABLE_HOMEASSISTANT I then have to enable MQTT and disable AMQTT. Enabling MQTT_HOME_ASSISTANT_SUPPORT requires AMQTT, which then conflicts with the requirement for ENABLE_HOMEASSISTANT.

Or am I missing something?

```
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT must be active
//#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
```

debsahu, 2018-06-25:
I'm sorry about the confusion: When `#define ENABLE_HOMEASSISTANT` is enabled, `#define ENABLE_MQTT` and `#define ENABLE_AMQTT` can be used interchangeably, but only one of them can be enabled. Secondly, I recommend `ENABLE_AMQTT` as its lightweight. When using `#define MQTT_HOME_ASSISTANT_SUPPORT` about 1.3kB of data is sent as the configuration for McLighting and PubSubClient/`ENABLE_MQTT` cannot handle that size data, but `ENABLE_AMQTT` can.

Regarding the pins, what is your schematic look like? Are you using the pin directly with NeoPixels or is it connected via a level shifter? D4/GPIO2 has no issues with NodeMCU, I know that it is connected to blue LED on ESP12E but does not cause any issues from the various testing I have done. Also could you use GPIO directly `#define PIN 2 // same as D4` in **definitions.h**?

SirJMD, 2018-06-30:
@debsahu Makes sense then! I'm currently using AMQTT and it works great.

The schematic is just like any other D1 mini (or clone). Right now I'm using it without a level shifter, since it's still in the mail somewhere. Although supplied with 5V, the D1 mini is 3.3V logic, right?

In definitions.h my current setting is **#define PIN 4** and it works without issues.

closed 2018-10-15 by debsahu