toblum/McLighting #190: Error compiling for board NodeMCU 1.0 (ESP-12E Module).

cjs1502, 2018-06-19:
when ever i try to compile the code and upload it to my nodemcu it shows this error code. and i am a newbie at this.please help me.

debsahu, 2018-06-19:
What is the error you are seeing?

cjs1502, 2018-06-19:
C:\Users\Dell\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp: In member function 'void WS2812FX::setPixelColor(uint16_t, uint32_t)':

C:\Users\Dell\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp:93:49: error: 'gamma8' was not declared in this scope

     Adafruit_NeoPixel::setPixelColor(n, gamma8(r), gamma8(g), gamma8(b), gamma8(w));

                                                 ^

C:\Users\Dell\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp: In member function 'void WS2812FX::setPixelColor(uint16_t, uint8_t, uint8_t, uint8_t)':

C:\Users\Dell\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp:101:49: error: 'gamma8' was not declared in this scope

     Adafruit_NeoPixel::setPixelColor(n, gamma8(r), gamma8(g), gamma8(b));

                                                 ^

C:\Users\Dell\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp: In member function 'void WS2812FX::setPixelColor(uint16_t, uint8_t, uint8_t, uint8_t, uint8_t)':

C:\Users\Dell\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp:109:49: error: 'gamma8' was not declared in this scope

     Adafruit_NeoPixel::setPixelColor(n, gamma8(r), gamma8(g), gamma8(b), gamma8(w));

                                                 ^

C:\Users\Dell\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_running_lights()':

C:\Users\Dell\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp:715:87: error: 'sine8' was not declared in this scope

  int lum = (int)sine8(((i + SEGMENT_RUNTIME.counter_mode_step) * 256) / SEGMENT_LENGTH);

                                                                                       ^

Multiple libraries were found for "Adafruit_NeoPixel.h"
 Used: C:\Users\Dell\Documents\Arduino\libraries\Adafruit_NeoPixel
 Not used: C:\Users\Dell\Documents\Arduino\libraries\Adafruit_NeoPixel-master
Multiple libraries were found for "WiFiManager.h"
 Used: C:\Users\Dell\Documents\Arduino\libraries\WiFiManager
 Not used: C:\Users\Dell\Documents\Arduino\libraries\WiFiManager-master
 Not used: C:\Users\Dell\Documents\Arduino\libraries\WhareHauoraWiFiManager
Multiple libraries were found for "WebSockets.h"
 Used: C:\Users\Dell\Documents\Arduino\libraries\WebSockets
 Not used: C:\Users\Dell\Documents\Arduino\libraries\arduinoWebSockets-master
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

debsahu, 2018-06-19:
Update your Adafruit NeoPixel library, gamma8 and sine8 were introduced recently.

debsahu, 2018-06-19:
At the bottom of your errors 
```
Multiple libraries were found for "WiFiManager.h"
Used: C:\Users\Dell\Documents\Arduino\libraries\WiFiManager
Not used: C:\Users\Dell\Documents\Arduino\libraries\WiFiManager-master
Not used: C:\Users\Dell\Documents\Arduino\libraries\WhareHauoraWiFiManager
Multiple libraries were found for "WebSockets.h"
Used: C:\Users\Dell\Documents\Arduino\libraries\WebSockets
Not used: C:\Users\Dell\Documents\Arduino\libraries\arduinoWebSockets-master
Multiple libraries were found for "Adafruit_NeoPixel.h"
Used: C:\Users\Dell\Documents\Arduino\libraries\Adafruit_NeoPixel
Not used: C:\Users\Dell\Documents\Arduino\libraries\Adafruit_NeoPixel-master
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```
you can see that you have multiple copies of WiFiManager, WebSockets and Adafruit_NeoPixel.

Delete the following folders
- C:\Users\Dell\Documents\Arduino\libraries\WiFiManager
- C:\Users\Dell\Documents\Arduino\libraries\WhareHauoraWiFiManager
- C:\Users\Dell\Documents\Arduino\libraries\WebSockets
- C:\Users\Dell\Documents\Arduino\libraries\Adafruit_NeoPixel

cjs1502, 2018-06-19:
OMG....!!!!!
It Worked.
you are the best
Thank you so much for your help and for you time,
i appreciate that.
thanks a lot

closed 2018-06-19 by debsahu

littlestud, 2019-02-20:
Please help me with this error :"(

`C:\Users\User\Documents\test\geolocation\geolocation.ino:2:25: fatal error: ArduinoJson.h: No such file or directory

 #include <ArduinoJson.h>

                         ^

compilation terminated.

Multiple libraries were found for "WiFiClient.h"
 Used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi
 Not used: C:\Users\User\Documents\Arduino\libraries\WiFi
 Not used: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi
 Not used: C:\Users\User\Documents\Arduino\libraries\WiFi
 Not used: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi
 Not used: C:\Users\User\Documents\Arduino\libraries\WiFi
 Not used: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi
 Not used: C:\Users\User\Documents\Arduino\libraries\WiFi
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).`

toblum, 2019-02-20:
@littlestud Are you sure that you are trying to compile the right Script? Geolocation.ino is no part of McLighting. 
Regards

littlestud, 2019-02-26:
> @littlestud Are you sure that you are trying to compile the right Script? Geolocation.ino is no part of McLighting.
> Regards

I am trying to track location using nodemcu without using GPS module. I followed this link https://circuitdigest.com/microcontroller-projects/how-to-track-location-with-nodemcu-using-google-map-api. Then, i saved the code file and name as geolocation. But then when i try to compile, this error pop out. Pls help me

debsahu, 2019-02-26:
@littlestud How is this related to McLighting?