toblum/McLighting #195: Help! error: 'DynamicJsonDocument' was not declared in this scope

maddogs, 2018-06-25:
Can anybody give me a pointer why I can't compile this sketch?

I've triple checked everything in the software installation guide, tried compiling on a VM fresh install of Windows 10 And I keep getting the same error.

It's bugging me as I had this working perfectly in november last year.

The full error is attached. Cheers1

[DynamicJsonDocument.txt](https://github.com/toblum/McLighting/files/2134453/DynamicJsonDocument.txt)

debsahu, 2018-06-25:
Please update ArduinoJSON to latest release version: 6.x.x https://github.com/bblanchon/ArduinoJson/releases. There are breaking changes in ArduinoJSON 5.x.x to 6.x.x and not cross-compatible!

maddogs, 2018-06-25:
Thanks! Worked first time.

closed 2018-06-25 by maddogs