toblum/McLighting #196: Probs with connection between Mqtt and NodeRed

Teckware, 2018-06-25:
I have my problems with the share of mqtt. Whenever I have everything running, with 5 lighting units, they lose their connection to the mqtt brocker after a certain amount of time (about 10 minutes). But otherwise I find this software code already fantastic.

debsahu, 2018-06-26:
Does each McLighting have unique hostname?

Teckware, 2018-06-26:
Jeep all ESP8266 have their own name and via the IP address are still all reachable. Only via my NodeRed site they are not reachable after some time. If I leave one of the ESP's on the serial port of the Adruino IDE I see that the MQTT connection disappears and the ESP after the program lines from the "definitions.h" here: #ifdef ENABLE_MQTT
  #define MQTT_MAX_PACKET_SIZE 512
  #define MQTT_MAX_RECONNECT_TRIES 4

4 attempts to reestablish the connection, and if that fails to take an unclear state.

debsahu, 2018-06-26:
I'm suspecting the MQTT server is having a hard time keeping up, does this issue happen if only one McLighting is connected. Do you have `#define ENABLE_AMQTT` enabled? Or are you using `#define ENABLE_MQTT`?

Teckware, 2018-06-26:
I use ENABLE_MQTT and now have about 7 weather ESP sensors that send via MQTT and now the Garden Lights ESP's which are to be controlled via Mqtt. And I still use the old code for the mc lighting.
*************************************
// Neopixel
#define PIN 2 // PIN (2 / D4) where neopixel / WS2811 strip is attached
#define NUMLEDS 8 // Number of leds in the strip
// #define BUILTIN_LED 2 // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
// #define BUTTON 4 // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

const char HOSTNAME [] = "Garden Light5"; // Friedly hostname

// #define HTTP_OTA // If defined, enable Added ESP8266HTTPUpdateServer
// # define ENABLE_OTA // If defined, enable Arduino OTA code.
#define ENABLE_MQTT // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
// # define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT must be active
// #define ENABLE_BUTTON // If defined, enable button handl
***********************************

debsahu, 2018-06-26:
Cant help you there, could you update to the latest code? The old code had an error that was introduced by me (i think) where the **mqtt_clientid** was manually set.

closed 2018-07-10 by Teckware