toblum/McLighting #197: 'DynamicJsonDocument' was not declared in this scope

cjs1502, 2018-06-27:
In file included from C:\Users\Dell\Downloads\McLighting-master\Arduino\McLighting\McLighting.ino:218:0:

sketch\request_handlers.h: In function 'String listStatusJSON()':

request_handlers.h:281: error: 'DynamicJsonDocument' was not declared in this scope

   DynamicJsonDocument jsonBuffer(bufferSize);

   ^

request_handlers.h:281: error: expected ';' before 'jsonBuffer'

   DynamicJsonDocument jsonBuffer(bufferSize);

                       ^

request_handlers.h:282: error: 'jsonBuffer' was not declared in this scope

   JsonObject& root = jsonBuffer.to<JsonObject>();

                      ^

request_handlers.h:282: error: expected primary-expression before '>' token

   JsonObject& root = jsonBuffer.to<JsonObject>();

                                              ^

request_handlers.h:282: error: expected primary-expression before ')' token

   JsonObject& root = jsonBuffer.to<JsonObject>();

                                                ^

request_handlers.h:294: error: 'serializeJson' was not declared in this scope

   serializeJson(root, json);

                           ^

sketch\request_handlers.h: In function 'String listModesJSON()':

request_handlers.h:306: error: 'DynamicJsonDocument' was not declared in this scope

   DynamicJsonDocument jsonBuffer(bufferSize);

   ^

request_handlers.h:306: error: expected ';' before 'jsonBuffer'

   DynamicJsonDocument jsonBuffer(bufferSize);

                       ^

request_handlers.h:307: error: 'jsonBuffer' was not declared in this scope

   JsonArray& json = jsonBuffer.to<JsonArray>();

                     ^

request_handlers.h:307: error: expected primary-expression before '>' token

   JsonArray& json = jsonBuffer.to<JsonArray>();

                                            ^

request_handlers.h:307: error: expected primary-expression before ')' token

   JsonArray& json = jsonBuffer.to<JsonArray>();

[... 311 lines cut ...]

     JsonObject& json = jsonBuffer.to<JsonObject>();

                        ^

McLighting:496: error: expected primary-expression before '>' token

     JsonObject& json = jsonBuffer.to<JsonObject>();

                                                ^

McLighting:496: error: expected primary-expression before ')' token

     JsonObject& json = jsonBuffer.to<JsonObject>();

                                                  ^

McLighting:553: error: 'serializeJson' was not declared in this scope

     serializeJson(json, json_str);

                                 ^

exit status 1
'DynamicJsonDocument' was not declared in this scope

closed 2018-06-27 by cjs1502

bagotu, 2018-07-08:
I have the same problem

debsahu, 2018-07-08:
@bagotu Please update ArduinoJSON to latest release version: 6.0.1-beta https://github.com/bblanchon/ArduinoJson/releases. There are breaking changes in ArduinoJSON 5.x.x to 6.x.x and not cross-compatible!

bagotu, 2018-07-08:
Yes of course ArduinoJson by @bblanchon (tested with 6.0.1). I used 6.1.0
now everything worked. Thank you